mkaczanowski/packer-plugin-builder-arm #5: Incompatible with Packer 1.5.x

ciarancourtney, 2020-02-03:
Getting error below:

    sudo packer build boards/beaglebone-black/debian.json 
    Failed to initialize build 'arm': error initializing builder 'arm': fork/exec /usr/bin: permission denied

Possibly related https://github.com/hashicorp/packer/issues/8527

mkaczanowski, 2020-02-04:
It's likely related to the issue upstream. I have manually update go.mod to point to packer 1.5.1 and the issue still persists.

Vagrant env used for test:
https://gist.github.com/mkaczanowski/be00b05adf2adcdaa9e89facf495285c

I am not going to investigate atm, but rather wait for the issue to be fixed upstream. For now you should use packer 1.4.x

skatsubo, 2020-02-25:
With Packer 1.5.4 there is more meaningful message:
`Failed to initialize build 'arm': error initializing builder 'arm': Incompatible API version with plugin. Plugin version: 4, Ours: 5`
As far as I understand, there is breaking API changes in 1.5.x vs 1.4.x.

(Thanks for the great project, btw!)

mkaczanowski, 2020-03-15:
Yeah, I saw there will be some changes required. Feel free to send PR, otherwise I'll look into this in a few weeks

mkaczanowski, 2020-04-02:
bump to 1.5: https://github.com/mkaczanowski/packer-builder-arm

closed 2020-04-02 by mkaczanowski