mkaczanowski/packer-plugin-builder-arm #8: Ability to use existing img

rmb938, 2020-03-10:
I have the following packer defintion

```
{
    "variables": {},
    "builders": [{
      "type": "arm",
      "file_urls" : ["https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip"],
      "file_checksum_url": "https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip.sha256",
      "file_checksum_type": "sha256",
      "file_target_extension": "zip",
      "image_build_method": "reuse",
      "image_path": "base.img",
      "image_size": "2G",
      "image_type": "dos",
      "image_partitions": [
        {
          "name": "boot",
          "type": "c",
          "start_sector": "8192",
          "filesystem": "vfat",
          "size": "256M",
          "mountpoint": "/boot"
        },
        {
          "name": "root",
          "type": "83",
          "start_sector": "532480",
          "filesystem": "ext4",
          "size": "0",
          "mountpoint": "/"
        }
      ],
      "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
      "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
      "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
    }],
    "provisioners": [
      {
        "type": "shell",
        "inline": [
          "apt update"
        ]
      }
    ]
  }
```

I want to be able to use the output image `base.img` in another packer run

mkaczanowski, 2020-03-10:
Hey,
I am not sure if I follow, but what do you mean by "another packer run"?

Here is an example where the output file is exported to docker:
https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/armv7/archlinuxarm-docker.json

is that what you're looking for?

rmb938, 2020-03-10:
Nope, let me try an explain a bit more.

If I have the packer configuration above and it outputs `base.img`

I would like to have another packer configuration something like what I have bellow. This one takes in `base.img` and outputs `vim.img` 

```
{
    "variables": {},
    "builders": [{
      "type": "arm",
      "existing_image" : "base.img",
      "image_build_method": "reuse",
      "image_path": "vim.img",
      "image_size": "2G",
      "image_type": "dos",
      "image_partitions": [
        {
          "name": "boot",
          "type": "c",
          "start_sector": "8192",
          "filesystem": "vfat",
          "size": "256M",
          "mountpoint": "/boot"
        },
        {
          "name": "root",
          "type": "83",
          "start_sector": "532480",
          "filesystem": "ext4",
          "size": "0",
          "mountpoint": "/"
        }
      ],
      "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
      "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
      "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
    }],
    "provisioners": [
      {
        "type": "shell",
        "inline": [
          "apt install vim"
        ]
      }
    ]
  }
```

rmb938, 2020-03-10:
Actually I seemed to have found a solution myself.

base.json
```
{
    "variables": {},
    "builders": [{
      "type": "arm",
      "file_urls" : ["https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip"],
      "file_checksum_url": "https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip.sha256",
      "file_checksum_type": "sha256",
      "file_target_extension": "zip",
      "image_build_method": "reuse",
      "image_path": "base.img",
      "image_size": "2G",
      "image_type": "dos",
      "image_partitions": [
        {
          "name": "boot",
          "type": "c",
          "start_sector": "8192",
          "filesystem": "vfat",
          "size": "256M",
          "mountpoint": "/boot"
        },
        {
          "name": "root",
          "type": "83",
          "start_sector": "532480",
          "filesystem": "ext4",
          "size": "0",
          "mountpoint": "/"
        }
      ],
      "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
      "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
      "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
    }],
    "provisioners": [
      {
        "type": "shell",
        "inline": [
          "apt update"
        ]
      }
    ],
    "post-processors": [
      {
        "type": "compress",
        "output": "base.img.zip"
      }
    ]
  }
```

`cloud-init.json`

```
{
    "variables": {},
    "builders": [{
      "type": "arm",
      "file_urls" : ["file:///images/base.img.zip"],
      "file_checksum_type": "none",
      "file_target_extension": "zip",
      "image_build_method": "reuse",
      "image_path": "cloud-init.img",
      "image_size": "2G",
      "image_type": "dos",
      "image_partitions": [
        {
          "name": "boot",
          "type": "c",
          "start_sector": "8192",
          "filesystem": "vfat",
          "size": "256M",
          "mountpoint": "/boot"
        },
        {
          "name": "root",
          "type": "83",
          "start_sector": "532480",
          "filesystem": "ext4",
          "size": "0",
          "mountpoint": "/"
        }
      ],
      "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
      "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
      "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
    }],
    "provisioners": [
      {
        "type": "shell",
        "inline": [
          "apt install -y cloud-init"
        ]
      }
    ]
  }
```

closed 2020-03-10 by rmb938

mkaczanowski, 2020-03-11:
yup, the go library that handles fetches also understand the local file "file://". Happy it worked for you