mkaczanowski/packer-plugin-builder-arm #12: feature request: support for unpacked base images

belkone, 2020-03-23:
Hi, as in the title, this would be a useful option for building images using previously prepared base images :)

mkaczanowski, 2020-03-24:
you can actually do it. If you use the "reuse" method, you need to specify previously create image to build ontop:
https://github.com/mkaczanowski/packer-builder-arm/blob/3cea7342db02c64f3ba98545b6edfce8b784d7d6/boards/beaglebone-black/debian.json#L10

is that what you're looking for

belkone, 2020-03-24:
Thanks for reply;) 
I think about possibility to set 'file_urls' to own prepared unpacked image. Right now it is possible but image must be packed (which extends time of all my workflow).
'reuse' is nice, but it replaces base image.
My workflow needs to build multiple images based on one special prepared image, which I don't want to change during build.

mkaczanowski, 2020-03-24:
I am parsing what you described. From what I understand your flow is as follows:
1. you have unpacked "rootfs" (aka unpaced base image) directory
2. you pass the "rootfs" to `file_urls` (or equivalent)
3. run chroot on the "rootfs"
4. build image (.img)

So what you can do now is to tarball the "rootfs" and pass it to packer. If we go your way with unpacked directory we would have to copy the files to "raw image" (img) anyway, so you only save the compression / decompression time.

Is that the flow you have in mind?

belkone, 2020-03-24:
I have multiple packer build instructions. 
In one case I have situation like this:
- builder 1 builds image_1.img, stores it as base for builder_2 and flashes it to SD cards for part of my RPIs.
- builder 2 is getting image_1.img as file_urls and rebuild it (expanded configuration / provisioning) as image_2.img which I can deploy to iscsi target or flash to sd. Right now I have to build img for flash and tarball it in post-process to use as file_urls in builder_2 instead of passing ready raw .img. And as you see I can't use 'reuse' option because I will lose image_1.img and I have additional job if I use rootfs instead of raw image which I need too.
builder_1 runs occasionally, but builder_2 is creating about 10 images at a time, well time saved it is very valuable ;)

Pozdrawiam ;)

mkaczanowski, 2020-03-24:
> Right now I have to build img for flash and tarball it in post-process to use
I don't think that's true, you should be able to pass raw image as input. ex. "file://image.img" have you tried that?

belkone, 2020-03-24:
Hmm.. if I use `"file_urls" : ["file://raspberry-pi.img"]` 
then build returns error:
`* couldn't extract checksum from checksum file`
but checksum file is ok.

if I use `"file_urls" : ["raspberry-pi.img"]` 
then it returns:
`==> arm: Retrieving rootfs_archive
==> arm: Trying raspberry-pi.img?archive=false
==> arm: Trying raspberry-pi.img?archive=false&checksum=md5%3Af5f9a57039e73e04ee3d29108d310c76
==> arm: raspberry-pi.img?archive=false&checksum=md5%3Af5f9a57039e73e04ee3d29108d310c76 => /build/packer_cache/ce7f2ebb48b311fe6d174d9621e95fb714877340
    arm: unpacking /build/packer_cache/ce7f2ebb48b311fe6d174d9621e95fb714877340 to raspbian-iscsi.img
==> arm: error while unpacking format unrecognized by filename: /build/packer_cache/ce7f2ebb48b311fe6d174d9621e95fb714877340: N/A
Build 'arm' errored: build was halted`

I tried to set 'file_target_extension' and 'file_unarchive_cmd ' to empty too but with no luck.

That's why I created this request - let me know if I did something wrong ;)

mkaczanowski, 2020-03-24:
see:
https://github.com/mkaczanowski/packer-builder-arm/issues/8

belkone, 2020-03-25:
Yeah, I saw it, but as I wrote it's not working if I use raw .img.:
`==> arm: Retrieving rootfs_archive
==> arm: Trying raspberry-pi.img?archive=false
==> arm: Trying raspberry-pi.img?archive=false
==> arm: raspberry-pi.img?archive=false => /build/packer_cache/aa2bae7ad1bf812100c8b6d2e7eb698489d21248.img
    arm: unpacking /build/packer_cache/aa2bae7ad1bf812100c8b6d2e7eb698489d21248.img to raspbian-iscsi.img
==> arm: error while unpacking format unrecognized by filename: /build/packer_cache/aa2bae7ad1bf812100c8b6d2e7eb698489d21248.img: N/A
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted
`
It works only if I zip or tar.gz this .img

belkone, 2020-03-25:
Maybe here it is possible to add a condition to support raw iso?

https://github.com/mkaczanowski/packer-builder-arm/blob/3cea7342db02c64f3ba98545b6edfce8b784d7d6/builder/step_extract_and_copy_image.go#L47-L56

mkaczanowski, 2020-03-25:
yeah, I think that's the place to plugin. PRs are welcome

belkone, 2020-03-28:
I'll try to look at it today or tomorrow

belkone, 2020-03-28:
@mkaczanowski I added the PR #14

closed 2020-03-30 by mkaczanowski