mkaczanowski/packer-plugin-builder-arm #16: Network access in chroot Arch?

bcomnes, 2020-04-02:
Im trying to see if I can manage to get network access in the chroot environment.  From what I can tell, this might be possible, but I keep running into this when I try to use pacman:

```
==> arm: + pacman -Sy --noconfirm --needed
613
    arm: :: Synchronizing package databases...
614
==> arm: error: failed retrieving file 'core.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
615
==> arm: error: failed to update core (invalid url for server)
616
==> arm: error: failed retrieving file 'extra.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
617
==> arm: error: failed to update extra (invalid url for server)
618
==> arm: error: failed retrieving file 'community.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
619
==> arm: error: failed to update community (invalid url for server)
620
==> arm: error: failed retrieving file 'alarm.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
621
==> arm: error: failed to update alarm (invalid url for server)
622
==> arm: error: failed retrieving file 'aur.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
623
==> arm: error: failed to update aur (invalid url for server)
624
==> arm: error: failed to synchronize all databases
```

Still trying to troubleshoot.  Sometimes Arch is picky about resolving when the clock is not set right to DNSSEC stuff.  If you have any idea or have experimented with this, pointers appreciated. I'll close when I resolve or give up.

bcomnes, 2020-04-02:
https://wiki.archlinux.org/index.php/Chroot#Using_chroot mentions copying over /etc/resolv.conf to the environment or using arch-chroot.  Is that possible with this builder?  I'll take a peek at the config code.

mkaczanowski, 2020-04-02:
Yes you need to update resolv.conf.

what I do is:
"echo 8.8.8.8 > /etc/resolv.conf"

whenever I need internet access

closed 2020-04-02 by mkaczanowski

bcomnes, 2020-04-02:
Running into an issue where I can't read/write to /etc or something:

```
==> arm: + touch /etc/resolv.conf
194
==> arm: /usr/sbin/touch: cannot touch '/etc/resolv.conf': No such file or directory
```

Listing `/`, I can see that `/etc` is right there.  Maybe its a permissions things.  Troubleshooting now.

bcomnes, 2020-04-02:
Got it working! https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/wandboard/archlinuxarm.json was a good reference here.