mkaczanowski/packer-plugin-builder-arm #18: File provisioner not working?

nazarewk, 2020-04-03:
I am trying to use file provisioner to put files into the image, but i'm getting the following error:
```
==> arm: Uploading Makefile => /tmp/Makefile
2020/04/03 13:25:08 Error in ProgressTrackingClient.Read RPC call: unexpected EOF
2020/04/03 13:25:08 Error in ProgressTrackingClient.Read RPC call: connection is shut down
2020/04/03 13:25:08 packer-provisioner-file plugin: [INFO] 1135 bytes written for 'uploadData'
2020/04/03 13:25:08 Error in Ui.Error RPC call: connection is shut down
2020/04/03 13:25:08 packer-provisioner-file plugin: closing
2020/04/03 13:25:08 closing
2020/04/03 13:25:08 Error in ProgressTrackingClient.Close RPC call: connection is shut down
2020/04/03 13:25:08 [INFO] (telemetry) ending file
==> arm: error while running provision hook: msgpack decode error [pos 1016]: cannot read container length: unrecognized descriptor byte: hex: c4, decimal: 196
Makefile 0 B / 1.11 KiB [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
2020/04/03 13:25:11 [INFO] (telemetry) ending arm
2020/04/03 13:25:11 machine readable: error-count []string{"1"}
==> Some builds didn't complete successfully and had errors:
2020/04/03 13:25:11 machine readable: arm,error []string{"build was halted"}
Build 'arm' errored: build was halted

==> Builds finished but no artifacts were created.
2020/04/03 13:25:11 [INFO] (telemetry) Finalizing.
==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

Makefile 0 B / 1.11 KiB [---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
2020/04/03 13:25:12 waiting for all plugin processes to complete...
2020/04/03 13:25:12 /home/e-kfni/dev/github.com/nazarewk-iac/packer-rpi-k3s/bin/packer: plugin process exited
2020/04/03 13:25:12 /home/e-kfni/dev/github.com/nazarewk-iac/packer-rpi-k3s/bin/packer-builder-arm: plugin process exited
2020/04/03 13:25:12 /home/e-kfni/dev/github.com/nazarewk-iac/packer-rpi-k3s/bin/packer: plugin process exited
2020/04/03 13:25:12 /home/e-kfni/dev/github.com/nazarewk-iac/packer-rpi-k3s/bin/packer: plugin process exited
make build  65.01s user 17.53s system 72% cpu 1:53.41 total
```

The code can be found here: https://github.com/nazarewk-iac/packer-rpi-k3s/tree/88f2fb70fe04e0adf33d89524178d68ae0a6e910

mkaczanowski, 2020-04-04:
There was something odd happening in go.mod (after a longer troubleshooting session...). Rebase on master and check again

closed 2020-04-04 by mkaczanowski

nazarewk, 2020-04-05:
works now, thanks!