mkaczanowski/packer-plugin-builder-arm #19: Sparse file support

bcomnes, 2020-04-05:
When creating large images, most CI environments top out at around 15-50GB of available space.  If you wanted to create a file system on the image for a 64gb card, this can't be done with this builder currently in these environments (you run out of space). 

Do you see any reason why a sparce file couldn't work here?  Going to test that out but curious if anyone has any thoughts on this.

dbast, 2020-04-05:
@bcomnes Nice idea... was thinking about the same and gave it a try here: https://github.com/mkaczanowski/packer-builder-arm/pull/20

mkaczanowski, 2020-04-05:
@bcomnes I am wondering why do you need such a big image anyway?

It is a common pattern to create a small partition for system image and resize it one the first run... this way you lower down the churn on your sdcard and speed up the whole process.

mkaczanowski, 2020-04-05:
merged @dbast pull request, that should do the job

closed 2020-04-05 by mkaczanowski

bcomnes, 2020-04-05:
HY! @mkaczanowski with packer 1.5 support and @dbast doing the hard work for me ❤️ I still have a Arm8/Raspi3/4 example I will PR once I finish up on my stuff.  This builder is awesome!  Works in all they ways I need it to. 

> It is a common pattern to create a small partition for system image and resize it one the first run... 

The default arrangement for arch installs is partition 1 on `/boot` and partition 2 on `/`.  To resize, you need to unmount the partition you resize, which I cannot figure out how to do on `/` while running from the system thats using it.  Any idea how to do that? This would make a helpful addition to the README.  The only way I could figure out how to do that was from a second linux  system and a USB reader, which kind of defeats the purpose of having a fully bootstrapped image ready to go that I can generate on my Mac with something like balena etcher.

mkaczanowski, 2020-04-05:
The way I do it:
1. run packer to generate the base image
2. in provisioned copy+enable systemd file:
```
[Unit]
Description=resize root file system
Before=local-fs-pre.target
DefaultDependencies=no

[Service]
Type=oneshot
TimeoutSec=infinity
ExecStart=/usr/sbin/resizerootfs
ExecStart=/bin/systemctl --no-reload disable %n

[Install]
RequiredBy=local-fs-pre.target
```

and this is the script:
```
#!/bin/sh

roottmp=$(lsblk -l -o NAME,MOUNTPOINT | grep '/$')
rootpart=/dev/${roottmp%% */}
rootdev=/dev/$(echo $roottmp | awk '{split($0,a,"p"); print a[1]}')
cnt=$(echo $roottmp | awk '{split($0,a,"p"); print a[2]}')

flock $rootdev sfdisk -f $rootdev -N $cnt <<EOF
,+
EOF

sleep 5

udevadm settle

sleep 5

flock $rootdev partprobe $rootdev

mount -o remount,rw $rootpart

resize2fs $rootpart

exit 0
```

What the script does is to resize the partition to max available space. The systemd file does it once (oneshoot) on the first boot, before the rootfs is mounted

bcomnes, 2020-04-05:
Super clever.  Thank you!  Want me to PR an example of this set up in the boards folder or add it to the README?

bcomnes, 2020-04-05:
@dbast your PR worked perfectly.  Glad that capability was added and it sounds like it has some additional benefits.  

Going to explore the resize solution @mkaczanowski mentioned because that is by far the most flexible solution.

bcomnes, 2020-04-06:
@mkaczanowski ran into an issue running that script:

```
resizerootfs.service - resize root file system
     Loaded: loaded (/etc/systemd/system/resizerootfs.service; disabled; vendor preset: disabled)
     Active: inactive (dead) since Thu 2020-02-13 20:01:20 UTC; 1 months 21 days ago
    Process: 369 ExecStart=/usr/sbin/resizerootfs (code=exited, status=0/SUCCESS)
    Process: 446 ExecStart=/bin/systemctl --no-reload disable resizerootfs.service (code=exited, status=0/SUCCESS)
   Main PID: 446 (code=exited, status=0/SUCCESS)

Feb 13 20:01:10 raspi3 resizerootfs[384]: Calling ioctl() to re-read partition table.
Feb 13 20:01:10 raspi3 resizerootfs[384]: Re-reading the partition table failed.: Device or resource busy
Feb 13 20:01:10 raspi3 resizerootfs[384]: The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or >
Feb 13 20:01:10 raspi3 resizerootfs[384]: Syncing disks.
Feb 13 20:01:20 raspi3 resizerootfs[442]: flock: failed to execute partprobe: No such file or directory
Feb 13 20:01:20 raspi3 resizerootfs[444]: resize2fs 1.45.5 (07-Jan-2020)
Feb 13 20:01:20 raspi3 resizerootfs[444]: The filesystem is already 422489 (4k) blocks long.  Nothing to do!
Feb 13 20:01:20 raspi3 systemctl[446]: Removed /etc/systemd/system/local-fs-pre.target.requires/resizerootfs.service.
Feb 13 20:01:20 raspi3 systemd[1]: resizerootfs.service: Succeeded.
Feb 13 20:01:20 raspi3 systemd[1]: Started resize root file system.
```

via: https://github.com/bcomnes/raspi-packer/blob/3d637bec338a2fd9238c865b82209621a3adde8e/resizerootfs#L3

Debugging now but basically the partition didn't resize it looks like.  Will post updates if I figure it out.

bcomnes, 2020-04-06:
Looks like re-enabling the service and restarting the system worked, but not sure if the way it was intended:

```
Feb 13 20:01:10 raspi3 resizerootfs[387]: The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or >
Feb 13 20:01:10 raspi3 resizerootfs[387]: Syncing disks.
Feb 13 20:01:21 raspi3 resizerootfs[434]: flock: failed to execute partprobe: No such file or directory
Feb 13 20:01:21 raspi3 resizerootfs[436]: resize2fs 1.45.5 (07-Jan-2020)
Feb 13 20:01:27 raspi3 resizerootfs[436]: Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
Feb 13 20:01:27 raspi3 resizerootfs[436]: old_desc_blocks = 1, new_desc_blocks = 8
Feb 13 20:01:27 raspi3 resizerootfs[436]: The filesystem on /dev/mmcblk0p2 is now 15580672 (4k) blocks long.
Feb 13 20:01:27 raspi3 systemctl[439]: Removed /etc/systemd/system/local-fs-pre.target.requires/resizerootfs.service.
Feb 13 20:01:27 raspi3 systemd[1]: resizerootfs.service: Succeeded.
Feb 13 20:01:27 raspi3 systemd[1]: Started resize root file system.
```

bcomnes, 2020-04-06:
Looks like I was missing parted.  Once that was installed it worked flawlessly.  What a great workflow, thank you for that!  I'll PR some more Arm8 examples for you to decide if you want them or not.

dbast, 2020-04-07:
A $script folder in the repo with e.g. resizerootfs + resizerootfs.service and one board definition using that as an example would be certainly nice.