mkaczanowski/packer-plugin-builder-arm #21: Can't resize partition during build

broose-goose, 2020-04-05:
I'm trying to run apt-get update/upgrade in the shell provisioner, but it returns that there isn't enough disk space. I tried setting the values of image_size, image_size_bytes, and to use the "size" option on the 2nd entry of image_partitions to no avail. Any thoughts?

mkaczanowski, 2020-04-05:
Can you paste the configuration?

Likely you don't have enough space... If you use 'reuse' option the image won't be "resized"

broose-goose, 2020-04-05:
Changing the option "image_build_method" from "reuse" to "new" solved that issue.

Now though, it hangs if I have any provisioners specified. For instance:

```
{
  "variables": {},
  "builders": [{
    "type": "arm",
    "file_urls" : ["https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip"],
    "file_checksum_url": "https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip.sha256",
    "file_checksum_type": "sha256",
    "file_target_extension": "zip",
    "image_build_method": "new",
    "image_path": "raspberry-pi.img",
    "image_size": "8G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "8192",
        "filesystem": "vfat",
        "size": "256M",
        "mountpoint": "/boot"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "532480",
        "filesystem": "ext4",
        "size": "0",
        "mountpoint": "/"
      }
    ],
    "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
    "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
  }],
  "provisioners": [
  ]
}
```

Works fine, while:
```
{
  "variables": {},
  "builders": [{
    "type": "arm",
    "file_urls" : ["https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip"],
    "file_checksum_url": "https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip.sha256",
    "file_checksum_type": "sha256",
    "file_target_extension": "zip",
    "image_build_method": "new",
    "image_path": "raspberry-pi.img",
    "image_size": "8G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "8192",
        "filesystem": "vfat",
        "size": "256M",
        "mountpoint": "/boot"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "532480",
        "filesystem": "ext4",
        "size": "0",
        "mountpoint": "/"
      }
    ],
    "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
    "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "touch /tmp/test"
      ]
    }
  ]
}
```

Gives the following output:

```
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842
==> arm: https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842 => /home/brucegoose/qemu_vms/raspbian/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip
    arm: creating an empty image raspberry-pi.img
==> arm: dd output: 0+0 records in
==> arm: 0+0 records out
==> arm: 0 bytes copied, 8.5671e-05 s, 0.0 kB/s
==> arm:
    arm: creating 2 dos partitions on raspberry-pi.img
    arm: searching for empty loop device (to map raspberry-pi.img)
    arm: mapping image raspberry-pi.img to /dev/loop20
    arm: running mkfs
    arm: mounting /dev/loop20p2 to /tmp/171083125
    arm: mounting /dev/loop20p1 to /tmp/171083125/boot
    arm: unpacking /home/brucegoose/qemu_vms/raspbian/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip to /tmp/171083125
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/171083125/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/171083125/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/171083125/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/171083125/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/171083125/sys]
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/qemu-arm
    arm: copying qemu binary from /usr/bin/qemu-arm-static to: /tmp/171083125/usr/bin/qemu-arm-static
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell060524661
==> arm: error while running provision hook: Error uploading script: exit status 1
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

broose-goose, 2020-04-05:
More information:

Running on Ubuntu 19.10 with 400G free. Go version 1.14.1

The "shell" provisioner worked fine when I had the setting "image_build_method": "reuse". Had to change this to resize the partition.

The "shell-local" provisioner works regardless of the setting "image_build_method"

mkaczanowski, 2020-04-05:
I think you're a bit lost here :) 

1. reuse - will fetch the image (.img) file, mount it and that's all. The `size` is ignored in reuse mode.
2. new - should only be used when you want to populate a roots (tarball with root file system).

What you're doing is the "reuse" mode. This is the image you're using:
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip

it's only 2G where the root partition is quite full "/" therefore you get "not enough space error"
```
    arm: Filesystem      Size  Used Avail Use% Mounted on
    arm: /dev/loop7p2    3.2G  2.7G  407M  87% /
    arm: /dev/loop7p1    253M   52M  201M  21% /boot
    arm: dev              12G     0   12G   0% /dev

```

this plugin won't support resizing base image, because it's complex and error prone business.

What I would suggest is to use other image, maybe with larger base partition

closed 2020-04-05 by mkaczanowski

mkaczanowski, 2020-04-26:
@broose-goose now you can resize partition:
https://github.com/mkaczanowski/packer-builder-arm#resizing-image