mkaczanowski/packer-plugin-builder-arm #29: Chroot & Reboot

eltucaso, 2020-04-18:
First of all, thank you for your awesome tool. I'm trying to build an advanced RPI image, but am running into issues. One I resolved myself, but think it may be worth your time to see how I resolved it in case it helps others. 

1. I followed the build instructions here: https://linuxhit.com/build-a-raspberry-pi-image-packer-packer-builder-arm/. On a fresh install of Ubuntu 18.04, after installing Go, Packer, & Packer-Builder-Arm, when I run the following command for a Raspberry PI Buster image:

 `sudo packer build xxxx.json`

the error below streams across the screen, distracting and hiding real errors from being read.  

`ERROR: ld.so: object /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.`

To remedy this I added the following to my .json via an inline command or to my build script: 

`mv /etc/ld.so.preload /etc/ld.so.preload.bak && touch /etc/ld.so.preload`

I'm sure other users are experiencing this issue, which is not unique to your tool, but also presented itself with the Solo-Io variant. I spent many hours trying to remedy this error, and gave up for a while and just let it stay since the image builds regardless, but of note, the above instruction was not found anywhere. The error still streams about 4 times during the initial build process, but doesn't stream incessantly. 

2. Like other users, I would like to resize a Raspberry Pi image to install software. I am aware that you do not want to get into the business of resizing base images, and have tried to use both your and the Raspberry built in /root filesystem expansion tools. Unfortunately, both tools require a reboot before they will work. When I try an inline command of:

`raspi-config --expand-rootfs && sleep 10 && reboot`

The expansion command appears to work after pressing the return key, as I get the following prompt in the status stream: 

`Please reboot`

But the reboot command does not work, instead I get:

`Running in chroot, ignoring request.`

Is there a way to force a reboot and continue on with the provisioning? If not, can I un-mount the image and remount without re-downloading the image again? I need a way to "reboot" the image AFTER download to allow it to resize itself. Thanks again for reading this and your fantastic efforts on an awesome tool, and I hope my questions help community.

mkaczanowski, 2020-04-18:
Hey,

### AD 1
This issue is related to the image you're using, ie. raspberry pi:
```
ERROR: ld.so: object /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
```

This is not the flaw of the tool, but rather the image itself. If you try the other images you might not see this issue at all.
I personally didn't investigate it, however, it's harmless enough to ignore it (afaik).

### AD 2
Resizing a partition is a complicated business. For example:
* some filesystems allow online resizing, other not
* the partition size vs image size are two different things. Say, you can have 2 Gb image where partitions in total take ~1.5Gb. What you can do (in theory) is to resize root partition by remaining ~500mb. Often this is not enough and users would like to resize the image (.img) as well but that is not possible without copying the files over to new-extended-size image.

It's a good practice to resize the partition on the first boot or so, like here:
https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi/archlinuxarm.json#L54

But I do see why users would like to resize the partitions in chroot... For now, it's not possible and maybe in a future me or someone else would implement the "copy method" as described above.

### Other notes
Running reboot within a container is not a good idea and as you see someone already thought about that:
```
Running in chroot, ignoring request
```

What would you expect to happen after reboot is issued? You don't want to reboot your PC (the one that runs the chroot and your operating system :) )

The approach you take is a bit off, what I think you could do:
1. move the resize partition to the first boot (see boards/raspberry-pi/archlinuxarm.json), for sure don't run this within the chroot
2. you might implement the "copy" method that would copy over the files from old image (ie. raspbian) to the new (custom created image with larger partitions). But be warned, there might be plenty of edge cases to think of
3. for sure don't run reboot in chroot. It's wrong and won't work, some of the operations that require the reboot you might want to move to ie. ansible and execute them on the first boot as well (or later, depends on use-case)
4. use a different image, the one with a larger root partition
5. instead of reusing images you might want to "init" the system image (ie. deb-bootstrap, pacinit...)

I think the first option is the easiest, so you might want to give it a try

closed 2020-04-18 by mkaczanowski

mkaczanowski, 2020-04-26:
@eltucaso now you can resize partitions if you want:
https://github.com/mkaczanowski/packer-builder-arm#resizing-image