mkaczanowski/packer-plugin-builder-arm #30: error while running provision hook: Error uploading script

lmfespinosa, 2020-04-19:
Hi! Im trying to make a packer image with a new one, and export the image base. The problem is when the script created the image, it can't provisioning (shell script, or shell command,...)

=> arm: https://images.offensive-security.com/arm-images/kali-linux-2020.1a-rpi3-nexmon.img.xz?archive=false&checksum=sha256%3A7ac77e7aadb70194791fc2463bc8fa413e88e7060740de53f1b29d79436303e7 => /home/espinosa/Escritorio/Cybersecurity-POCs-Automation/Raspberry Pi/Script_Packer_ARM_Rasp/packer_cache/b22dc5e67986bf41432c648da1f618afbef3a23b.xz
    arm: creating an empty image raspberry-pi-kali2.img
==> arm: dd output: 0+0 registros leídos
==> arm: 0+0 registros escritos
==> arm: 0 bytes copied, 0,000173755 s, 0,0 kB/s
==> arm:
    arm: creating 2 dos partitions on raspberry-pi-kali2.img
    arm: searching for empty loop device (to map raspberry-pi-kali2.img)
    arm: mapping image raspberry-pi-kali2.img to /dev/loop26
    arm: running mkfs
    arm: mounting /dev/loop26p2 to /tmp/745650543
    arm: mounting /dev/loop26p1 to /tmp/745650543/boot
    arm: unpacking /home/espinosa/Escritorio/Cybersecurity-POCs-Automation/Raspberry Pi/Script_Packer_ARM_Rasp/packer_cache/b22dc5e67986bf41432c648da1f618afbef3a23b.xz to /tmp/745650543
    arm: unpacking with custom comand: [xz -d /home/espinosa/Escritorio/Cybersecurity-POCs-Automation/Raspberry Pi/Script_Packer_ARM_Rasp/packer_cache/b22dc5e67986bf41432c648da1f618afbef3a23b.xz]
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/745650543/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/745650543/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/745650543/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/745650543/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/745650543/sys]
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/qemu-arm
    arm: copying qemu binary from /usr/bin/qemu-arm-static to: /tmp/745650543/usr/bin/qemu-arm-static
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell902857544
==> arm: error while running provision hook: Error uploading script: exit status 1
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.

When i use "reuse" works good, but my problem is that i have to resize the partition because i only have 29MB available on the original size.

The script is this:

"variables": {
    },
    "builders": [
      {
        "type": "arm",
        "file_urls": ["https://images.offensive-security.com/arm-images/kali-linux-2020.1a-rpi3-nexmon.img.xz"],
        "file_checksum_type": "sha256",
        "file_checksum": "7ac77e7aadb70194791fc2463bc8fa413e88e7060740de53f1b29d79436303e7",
        "image_build_method": "new",
        "file_target_extension": "xz",
        "file_unarchive_cmd": ["xz","-d", "$ARCHIVE_PATH"],
        "image_path": "raspberry-pi-kali2.img",
        "image_size": "10G",
        "image_type": "dos",
        "image_partitions": [
          {
            "name": "boot",
            "type": "c",
            "start_sector": "1",
            "filesystem": "vfat",
            "size": "121M",
            "mountpoint": "/boot"
          },
          {
            "name": "root",
            "type": "83",
            "start_sector": "250001",
            "filesystem": "ext4",
            "size": "0",
            "mountpoint": "/"
          }
        ],
        "image_setup_extra": [],
        "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
        "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
        "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
      }
    ],
    "provisioners": [
      
      {
        "type": "shell",
        "inline": [
          "apt-get update"
        ]
      }
    ]
}

Best regards

mkaczanowski, 2020-04-20:
Please see (threads about resize):
https://github.com/mkaczanowski/packer-builder-arm/issues/21
https://github.com/mkaczanowski/packer-builder-arm/issues/29

> Hi! Im trying to make a packer image with a new one, and export the image base. The problem is when the script created the image, it can't provisioning (shell script, or shell command,...)

Provisioner fails because you created empty image (no rootfs is being copied)

closed 2020-04-20 by mkaczanowski

lmfespinosa, 2020-04-20:
Thanks! 
I did the restore img to the new one. Now the result image hace the correct format, but the script finished with an error:
Build 'arm' errored: unexpected EOF

I did in the file_unarchive_cmd, a restoration image, because this image i need a chroot mode for resize in first boot mode.

The command works correctly for the restore partition in console.

Thanks in advance.