mkaczanowski/packer-plugin-builder-arm #34: arm: exit status 32 error

tanchtech, 2020-04-27:
Hi,

I encountered this error when trying out the docker commands in the readme. I'm running in a vagrant box.

`sudo docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm build boards/raspberry-pi/raspbian.json -extra-system-packages=bmap-tools,zstd`

The console output here:

```
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842
==> arm: https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842 => /build/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip
    arm: unpacking /build/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip to raspberry-pi.img
    arm: searching for empty loop device (to map raspberry-pi.img)
    arm: mapping image raspberry-pi.img to /dev/loop2
    arm: mounting /dev/loop2p2 to /tmp/829575546
==> arm: exit status 32
==> arm: exit status 32
==> arm: exit status 32
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

mkaczanowski, 2020-04-27:
@dbast do you want to take a look at this? :) (thanks!)

dbast, 2020-04-28:
No idea... the Docker container works here on Mac and Linux (also on Travis-CI).

@tanchtech: What OS/Linux kernel (e.g. `uname -a`) and and docker version (`docker -v`) are you using?

tanchtech, 2020-04-29:
Here are the versions:
- Windows 10
- Virtualbox 5.2.34
- Vagrant 2.2.7
- This repo, commit 94d23238

```ruby
# Vagrantfile
Vagrant.configure("2") do |config|
  config.vagrant.plugins = [
    "vagrant-vbguest",
    "vagrant-disksize",
  ]
  config.vbguest.auto_update = true

  config.vm.box = "ubuntu/bionic64"
  config.vm.box_version = "20200416.0.0"
  config.vm.network "public_network"
  config.disksize.size = "40GB"

  config.vm.provider "virtualbox" do |v|
    v.customize ["modifyvm", :id, "--nictype1", "virtio"]
    v.memory = 4096
  end

  config.vm.provision "ansible_local" do |ansible|
    ansible.galaxy_role_file = "requirements.yml"
    ansible.playbook = "playbook.yml"
    ansible.extra_vars = { ansible_python_interpreter: "/usr/bin/python3" }
  end
end
```

I used Ansible for some convenience in setting up.

```yml
# requirements.yml
---
- src: geerlingguy.docker
```

```yml
# playbook.yml
---
- hosts: all
  become: yes

  roles:
    - role: geerlingguy.docker
```

```bash
$ uname -a
Linux ubuntu-bionic 4.15.0-96-generic #97-Ubuntu SMP Wed Apr 1 03:25:46 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
$ docker -v
Docker version 19.03.8, build afacb8b7f0
```

dbast, 2020-04-30:
Uh, lots of layers ... Maybe via Windows -> Linux -> Docker -> qemu-binfmt some CPU features are lost due to nested virtualization and qemu-binfmt stops working. But that is just a guess... How can that be debugged? Comparing features in `/proc/cpuinfo` on different machines/layers?

tanchtech, 2020-05-05:
I figured out that the error was occuring in the mapped `/vagrant` directory in VirtualBox. The docker command works fine if I cloned this repo to another non-mapped directory (i.e. `~/`). So, it could be due to Windows->VirtualBox directory mapping issues.

closed 2020-05-05 by mkaczanowski