mkaczanowski/packer-plugin-builder-arm #36: Resize

eltucaso, 2020-05-11:
Guys,

Thanks a bunch for adding the resizing option, I prefer your build to Solo-IO and now I have the capabilities I need to move on. If it is helpful, here are my install steps to make this work on Ubuntu 18.04 in my install script: 

# Install Packer Builder Arm Mkaczanowski
`sudo apt-get install -y e2fsprogs kpartx qemu-utils parted
cd ~ && git clone https://github.com/mkaczanowski/packer-builder-arm
cd ~/packer-builder-arm && go mod download && go build`

The new resizing option appears to work great, but I'm getting an error at the completion of my builds: 
`optional fuser -k failed with exit status 1:`

I've tried reinstalling your tool with different methods on fresh Ubuntu 18.04 builds to fix the error to no avail. I'm not sure if the error is relevant or can just be ignored, Tomorrow I'll make sure the image boots without issue, though it looks like a good build. I haven't found anything on that error, so I am unable to troubleshoot. Thanks again.

mkaczanowski, 2020-05-11:
`fuser -k` kills everything that uses a mountopint, otherwise unmount would fail.

The fact it fails is strange. Can you check if it's installed?
```
which fuser

or install it
sudo apt-get install -y psmisc
```

eltucaso, 2020-05-11:
It is installed, and reports back at:
which user
/bin/fuser

mkaczanowski, 2020-05-11:
Can you repro this in some reproducible environment like this?:
https://gist.github.com/mkaczanowski/c77b03323d9f376e75c9e91063213b36

Try the vagrant config I gave you, change the things that are specific to your environment and let me know if you can repro this.

I am 99% sure something is wrong with your environment

eltucaso, 2020-05-11:
I tried the vagrant setting in my VirtualBox Ubunt setup, and ran into the
following issue:

   - Go: I'm installing Go manually, as the golang version is 1.10, which
   doesn't support go mod or go download
   - Packer- I'm installing version 1.5.5, as previous versions don't have
   the right plugin options

See the below commands from their respective repositories; I reboot after
Go install and after Packer Builder Arm install:

# Install Go
cd ~ && sudo wget -O ~/go1.13.5.linux-amd64.tar.gz
https://dl.google.com/go/go1.13.5.linux-amd64.tar.gz
sudo tar -C /usr/local -xvzf go1.13.5.linux-amd64.tar.gz
sudo echo "export PATH=$PATH:/usr/local/go/bin" >> ~/.profile

#Install Packer
cd ~/ && sudo apt-get install -y git unzip qemu-user-static e2fsprogs
dosfstools bsdtar
wget
https://releases.hashicorp.com/packer/1.5.5/packer_1.5.5_linux_amd64.zip
unzip packer_1.5.5_linux_amd64.zip
sudo mv packer /usr/local/bin/
export PATH=$PATH:/usr/local/bin

# Install Packer Builder Arm  Mkaczanowski
sudo apt-get install -y e2fsprogs kpartx qemu-utils parted
cd ~ && git clone https://github.com/mkaczanowski/packer-builder-arm
cd ~/packer-builder-arm && go mod download && go build

mkaczanowski, 2020-05-12:
This is the repo vagrant setup:
```
# -*- mode: ruby -*-
# vi: set ft=ruby :

# All Vagrant configuration is done below. The "2" in Vagrant.configure
# configures the configuration version (we support older styles for
# backwards compatibility). Please don't change it unless you know what
# you're doing.
Vagrant.configure("2") do |config|
  config.vm.box = "ubuntu/bionic64"
  config.disksize.size = '15GB'

  config.vm.provider "virtualbox" do |v|
      v.memory = 4096
      v.cpus = 2
  end

  config.vm.provision "shell", inline: <<-SHELL
    yes | sudo add-apt-repository ppa:longsleep/golang-backports

    apt-get update
    apt-get install -y git golang-go qemu-user-static unzip

	# install bsdtar
    wget https://www.libarchive.org/downloads/libarchive-3.3.2.tar.gz
    tar xzf libarchive-3.3.2.tar.gz
    cd libarchive-3.3.2
    ./configure
    make
    sudo make install

    cd ~/
    wget https://github.com/hashicorp/packer/archive/v1.5.5.zip
    unzip v1.5.5.zip
    cd packer-1.5.5
    go mod download
    go build -o /tmp/packer
    sudo cp -r /tmp/packer /usr/bin/

    cd ~/
    git clone https://github.com/mkaczanowski/packer-builder-arm
    cd packer-builder-arm
    go mod download
    go build

    sudo packer build boards/odroid-xu4/archlinuxarm.json
    sudo fdisk -l odroid-xu4.img
  SHELL
end
```

I tested it locally (via `vagrant up`) and all works fine, which is a clear indication that the problem is with your setup. What's wrong I can't tell, I don't have access to your PC.

I am closing this issue since you can't provide the repro and the Virtualbox setup with standard ubuntu 18.04 does work

closed 2020-05-12 by mkaczanowski