mkaczanowski/packer-plugin-builder-arm #38: Working example for Alpine Linux

dirkschneemann, 2020-06-12:
Hello,

I am trying to build an image of [Alpine Linux](https://alpinelinux.org/downloads/) for the Raspberry Pi (in my case Raspberry Pi Zero W). According to [this guide](https://wiki.alpinelinux.org/wiki/Raspberry_Pi_Zero_W_-_Installation), the partition layout is just a single FAT32 partition on which the content of the downloaded archive is extracted.

My minimal example packer build definition looks like this:

```
{
    "builders": [
        {
            "type": "arm",
            "file_urls": ["http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-rpi-3.12.0-armhf.tar.gz"],
            "file_checksum_type": "sha256",
            "file_checksum_url": "http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-rpi-3.12.0-armhf.tar.gz.sha256",
            "file_unarchive_cmd": ["tar", "-xzvf", "$ARCHIVE_PATH", "-C", "$MOUNTPOINT", "--no-same-owner"],
            "image_build_method": "new",
            "image_path": "alpine.iso",
            "image_size": "4G",
            "image_type": "dos",
            "image_partitions": [
                {
                    "name": "root",
                    "type": "b",
                    "start_sector": "0",
                    "filesystem": "vfat",
                    "size": "0",
                    "mountpoint": "/"
                }
            ]
        }
    ],
    "provisioners": [
        {
            "type": "shell",
            "inline_shebang": "/bin/sh -ex",
            "inline": [
                "echo 'Hello World'"
            ]
        }
    ]
}
```

However, this fails with the following error:
```
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-rpi-3.12.0-armhf.tar.gz?archive=false
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-rpi-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3Ae6df41d25e6afed68b50dc94d572f3fe8f0704802f0e9156cd6b571f72615009
==> arm: http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-rpi-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3Ae6df41d25e6afed68b50dc94d572f3fe8f0704802f0e9156cd6b571f72615009 => /home/schneema/Documents/raspberry-pi/packer_cache/2021f82067444ccfa151edecccf621e75b949e89
    arm: creating an empty image alpine.iso
==> arm: dd output: 0+0 records in
==> arm: 0+0 records out
==> arm: 0 bytes copied, 8.6662e-05 s, 0.0 kB/s
==> arm:
    arm: creating 1 dos partitions on alpine.iso
    arm: searching for empty loop device (to map alpine.iso)
    arm: mapping image alpine.iso to /dev/loop23
    arm: running mkfs
    arm: mounting /dev/loop23p1 to /tmp/156452692
    arm: unpacking /home/schneema/Documents/raspberry-pi/packer_cache/2021f82067444ccfa151edecccf621e75b949e89 to /tmp/156452692
    arm: unpacking with custom comand: [tar -xzvf /home/schneema/Documents/raspberry-pi/packer_cache/2021f82067444ccfa151edecccf621e75b949e89 -C /tmp/156452692 --no-same-owner]
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/156452692/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/156452692/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/156452692/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/156452692/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/156452692/sys]
==> arm: Failed to find binfmt_misc for  under /proc/sys/fs/binfmt_misc
==> arm: remove /tmp/156452692: device or resource busy
==> arm: optional `fuser -k` failed with exit status 1:
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

Any ideas what is configured incorrectly? Would someone have a working example for building an image based on Alpine Linux?

Cheers

mkaczanowski, 2020-06-12:
I don't know much about how alpine is designed, but... for sure it doesn't use standard Linux files-system-layout. See:
```
[admin@laptop alpine-rpi-3.12.0-armhf]$ ls -la 
total 6300
drwxr-xr-x  5 admin sudo     4096 May 29 16:21 .
drwx------ 47 admin admin   36864 Jun 12 11:38 ..
-rw-r--r--  1 admin sudo       25 May 29 16:21 .alpine-release
drwxr-xr-x  3 admin sudo     4096 May 29 16:21 apks
-rw-r--r--  1 admin sudo    25081 May 28 11:52 bcm2708-rpi-b.dtb
-rw-r--r--  1 admin sudo    25344 May 28 11:52 bcm2708-rpi-b-plus.dtb
-rw-r--r--  1 admin sudo    24692 May 28 11:52 bcm2708-rpi-b-rev1.dtb
-rw-r--r--  1 admin sudo    24862 May 28 11:52 bcm2708-rpi-cm.dtb
-rw-r--r--  1 admin sudo    24786 May 28 11:52 bcm2708-rpi-zero.dtb
-rw-r--r--  1 admin sudo    25522 May 28 11:52 bcm2708-rpi-zero-w.dtb
-rw-r--r--  1 admin sudo    26347 May 28 11:52 bcm2709-rpi-2-b.dtb
-rw-r--r--  1 admin sudo    26496 May 28 11:52 bcm2710-rpi-2-b.dtb
-rw-r--r--  1 admin sudo    27537 May 28 11:52 bcm2710-rpi-3-b.dtb
-rw-r--r--  1 admin sudo    28156 May 28 11:52 bcm2710-rpi-3-b-plus.dtb
-rw-r--r--  1 admin sudo    26351 May 28 11:52 bcm2710-rpi-cm3.dtb
-rw-r--r--  1 admin sudo    46412 May 28 11:52 bcm2711-rpi-4-b.dtb
-rw-r--r--  1 admin sudo    18262 May 28 11:52 bcm2835-rpi-a.dtb
-rw-r--r--  1 admin sudo    18382 May 28 11:52 bcm2835-rpi-a-plus.dtb
-rw-r--r--  1 admin sudo    18408 May 28 11:52 bcm2835-rpi-b.dtb
-rw-r--r--  1 admin sudo    18669 May 28 11:52 bcm2835-rpi-b-plus.dtb
-rw-r--r--  1 admin sudo    18541 May 28 11:52 bcm2835-rpi-b-rev2.dtb
-rw-r--r--  1 admin sudo    18523 May 28 11:52 bcm2835-rpi-cm1-io1.dtb
-rw-r--r--  1 admin sudo    18334 May 28 11:52 bcm2835-rpi-zero.dtb
-rw-r--r--  1 admin sudo    18777 May 28 11:52 bcm2835-rpi-zero-w.dtb
-rw-r--r--  1 admin sudo    19661 May 28 11:52 bcm2836-rpi-2-b.dtb
-rw-r--r--  1 admin sudo    19904 May 28 11:52 bcm2837-rpi-3-a-plus.dtb
-rw-r--r--  1 admin sudo    20353 May 28 11:52 bcm2837-rpi-3-b.dtb
-rw-r--r--  1 admin sudo    20773 May 28 11:52 bcm2837-rpi-3-b-plus.dtb
-rw-r--r--  1 admin sudo    19680 May 28 11:52 bcm2837-rpi-cm3-io3.dtb
drwxr-xr-x  2 admin sudo     4096 May 29 16:21 boot
-rwxr-xr-x  1 admin sudo    52304 Mar 10 11:38 bootcode.bin
-rw-r--r--  1 admin sudo       60 May 29 16:21 cmdline.txt
-rw-r--r--  1 admin sudo      565 May 29 16:21 config.txt
-rwxr-xr-x  1 admin sudo     6200 Mar 10 11:38 fixup4.dat
-rwxr-xr-x  1 admin sudo     6745 Mar 10 11:38 fixup.dat
drwxr-xr-x  2 admin sudo    12288 May 29 16:21 overlays
-rwxr-xr-x  1 admin sudo  2784800 Mar 10 11:38 start4.elf
-rwxr-xr-x  1 admin sudo  2883204 Mar 10 11:38 start.elf
```

Likely bootloader unpacks some rootfs filesystem onfly somewhere. Packer-builder-arm (or any other imaging application) won't work (aka you won't be able to change the base image in "standard way") because alpine comes with very much uniqiue/customized image

dirkschneemann, 2020-06-12:
That's too bad. Thanks for the quick reply though.

dirkschneemann, 2020-06-12:
One more question: I just discovered that Alpine is also [published](https://alpinelinux.org/downloads/) as `Minimal root filesystem. For use in containers and minimal chroots.`. Unpacking these archives looks much more like a regular Linux file system.

Applying this in my minimal example, and changing `vfat` to `ext4` since the archive uses symlinks which are not supported on FAT:
```
{
    "builders": [
        {
            "type": "arm",
            "file_urls": ["http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz"],
            "file_checksum_type": "sha256",
            "file_checksum_url": "http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz.sha256",
            "file_target_extension": "tar.gz",
            "image_build_method": "new",
            "image_path": "alpine.iso",
            "image_size": "4G",
            "image_type": "dos",
            "image_partitions": [
                {
                    "name": "root",
                    "type": "b",
                    "start_sector": "0",
                    "filesystem": "ext4",
                    "size": "0",
                    "mountpoint": "/"
                }
            ]
        }
    ],
    "provisioners": [
        {
            "type": "shell",
            "inline_shebang": "/bin/sh -ex",
            "inline": [
                "echo 'Hello World'"
            ]
        }
    ]
}
```

The error remains the same though:
```
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3A19d9d801fc121dc1c24aebdd7b73cad024adc26ef7d902497bd9c231fac6c0fe
==> arm: http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3A19d9d801fc121dc1c24aebdd7b73cad024adc26ef7d902497bd9c231fac6c0fe => /home/schneema/Documents/raspberry-pi/packer_cache/46d8c204e8f4cf71d3ea627a97a2f82042947390.tar.gz
    arm: creating an empty image alpine.iso
==> arm: dd output: 0+0 records in
==> arm: 0+0 records out
==> arm: 0 bytes copied, 0.000129546 s, 0.0 kB/s
==> arm:
    arm: creating 1 dos partitions on alpine.iso
    arm: searching for empty loop device (to map alpine.iso)
    arm: mapping image alpine.iso to /dev/loop23
    arm: running mkfs
    arm: mounting /dev/loop23p1 to /tmp/010546804
    arm: unpacking /home/schneema/Documents/raspberry-pi/packer_cache/46d8c204e8f4cf71d3ea627a97a2f82042947390.tar.gz to /tmp/010546804
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/010546804/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/010546804/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/010546804/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/010546804/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/010546804/sys]
==> arm: Failed to find binfmt_misc for  under /proc/sys/fs/binfmt_misc
==> arm: remove /tmp/010546804: device or resource busy
==> arm: optional `fuser -k` failed with exit status 1:
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

Isn't this archive format supported either?

mkaczanowski, 2020-06-18:
I'll check that in a while (a bit busy at the time, sorry)

jameseck, 2020-06-24:
I'm seeing the same fuser error trying to build with the repo provided raspbian.json file.  (I added the /etc vol since I was also getting an error about /etc/mtab not existing and hoped that would solve the fuser error - but it didn't).

This is on a Fedora 32 laptop via podman.

```
$ sudo podman run  --privileged -v /etc:/etc -v /dev:/dev -v ${PWD}:/build docker.io/mkaczanowski/packer-builder-arm build boards/raspberry-pi/raspbian.json
running /bin/packer build boards/raspberry-pi/raspbian.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false
==> arm: Trying https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842
==> arm: https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip?archive=false&checksum=sha256%3A2c4067d59acf891b7aa1683cb1918da78d76d2552c02749148d175fa7f766842 => /build/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip
    arm: unpacking /build/packer_cache/665ae48e54b8fac4541688233efc4a82b75de3ce.zip to raspberry-pi.img
    arm: searching for empty loop device (to map raspberry-pi.img)
    arm: mapping image raspberry-pi.img to /dev/loop0
    arm: mounting /dev/loop0p2 to /tmp/541670624
    arm: mounting /dev/loop0p1 to /tmp/541670624/boot
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/541670624/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/541670624/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/541670624/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/541670624/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/541670624/sys]
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/qemu-arm
    arm: copying qemu binary from /usr/bin/qemu-arm-static to: /tmp/541670624/usr/bin/qemu-arm-static
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell735960697
==> arm: optional `fuser -k` failed with exit status 1:
Build 'arm' finished.
```

mkaczanowski, 2020-06-24:
> optional `fuser -k` failed with exit status 1:

it's optional, you can ignore the error. To what I see your build finished fine?

mkaczanowski, 2020-06-26:
This diff brings the alpine configuration: https://github.com/mkaczanowski/packer-builder-arm/commit/4a35b8997f1c806a3281228d4de1ac0cf8ab63a2

Here is the proof it works :) The last issue you have reported likely is caused by the wrong qemu setup. Please try again with the command below (after git pull)
```
$ sudo packer build  boards/armv7/alpine.json

==> arm: Retrieving rootfs_archive
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false
==> arm: Trying http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3A19d9d801fc121dc1c24aebdd7b73cad024adc26ef7d902497bd9c231fac6c0fe
==> arm: http://dl-cdn.alpinelinux.org/alpine/v3.12/releases/armhf/alpine-minirootfs-3.12.0-armhf.tar.gz?archive=false&checksum=sha256%3A19d9d801fc121dc1c24aebdd7b73cad024adc26ef7d902497bd9c231fac6c0fe => /home/admin/Projects/packer-builder-arm/packer_cache/46d8c204e8f4cf71d3ea627a97a2f82042947390.tar.gz
    arm: creating an empty image alpine.img
==> arm: dd output: 0+0 records in
==> arm: 0+0 records out
==> arm: 0 bytes copied, 8.7848e-05 s, 0.0 kB/s
==> arm:
    arm: creating 1 dos partitions on alpine.img
    arm: searching for empty loop device (to map alpine.img)
    arm: mapping image alpine.img to /dev/loop3
    arm: running mkfs
    arm: mounting /dev/loop3p1 to /tmp/156588818
    arm: unpacking /home/admin/Projects/packer-builder-arm/packer_cache/46d8c204e8f4cf71d3ea627a97a2f82042947390.tar.gz to /tmp/156588818
    arm: unpacking with custom comand: [bsdtar -xpf /home/admin/Projects/packer-builder-arm/packer_cache/46d8c204e8f4cf71d3ea627a97a2f82042947390.tar.gz -C /tmp/156588818]
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/156588818/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/156588818/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/156588818/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/156588818/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/156588818/sys]
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/arm
    arm: copying qemu binary from /usr/bin/qemu-arm-static to: /tmp/156588818/usr/bin/qemu-arm-static
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell713038477
    arm: fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/main/armhf/APKINDEX.tar.gz
    arm: fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/community/armhf/APKINDEX.tar.gz
    arm: v3.12.0-94-g0ddce5a52e [http://dl-cdn.alpinelinux.org/alpine/v3.12/main]
    arm: v3.12.0-93-g291c4d97ff [http://dl-cdn.alpinelinux.org/alpine/v3.12/community]
    arm: OK: 11226 distinct packages available
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' finished.

==> Builds finished. The artifacts of successful builds are:
--> arm: alpine.img
```

closed 2020-06-26 by mkaczanowski