mkaczanowski/packer-plugin-builder-arm #39: unpacked base image option still attempts to unpack image

tannerjfco, 2020-07-07:
Per docs for unpacked base image:

> Raw images format (`.img` or `.iso`) can be used by defining the `file_target_extension` appropriately.

And addition of feature added in #14, I am trying to run a packer build using local iso of ubuntu 20.04 arm

I am using the following packer json to attempt initial build

```
{
  "variables": {},
  "builders": [{
    "type": "arm",
    "file_urls" : ["file:///build/ubuntu-20.04-live-server-arm64.iso"],
    "file_checksum_type": "none",
    "file_target_extension": "iso",
    "image_build_method": "new",
    "image_path": "raspberry-pi-4.img",
    "image_size": "2G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "2048",
        "filesystem": "vfat",
        "size": "256M",
        "mountpoint": "/boot"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "526336",
        "filesystem": "ext4",
        "size": "0",
        "mountpoint": "/"
      }
    ],
    "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "apt-get update && apt-get install -y golang"
      ]
    },
    {
      "type": "file",
      "source": "scripts/resizerootfs",
      "destination": "/tmp"
    },
    {
      "type": "shell",
      "script": "scripts/bootstrap_resizerootfs.sh"
    }
  ]
}
```
However, when running this build I get the following result

```
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm:local build boards/raspberry-pi-4/ubuntu.json
running /bin/packer build boards/raspberry-pi-4/ubuntu.json
arm: output will be in this color.

Warnings for build 'arm':

* a checksum type of 'none' was specified. Specyfying checksum is highly recommended
* a checksum type of 'none' was specified. Specyfying checksum is highly recommended

==> arm: Retrieving rootfs_archive
==> arm: Trying file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false
==> arm: Trying file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false
==> arm: file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false => /build/ubuntu-20.04-live-server-arm64.iso
    arm: creating an empty image raspberry-pi-4.img
==> arm: dd output: 0+0 records in
==> arm: 0+0 records out
==> arm: 0 bytes copied, 3.179e-05 s, 0.0 kB/s
==> arm:
    arm: creating 2 dos partitions on raspberry-pi-4.img
    arm: searching for empty loop device (to map raspberry-pi-4.img)
    arm: mapping image raspberry-pi-4.img to /dev/loop24
    arm: running mkfs
    arm: mounting /dev/loop24p2 to /tmp/789904072
    arm: mounting /dev/loop24p1 to /tmp/789904072/boot
    arm: unpacking /build/ubuntu-20.04-live-server-arm64.iso to /tmp/789904072
==> arm: error while unpacking format unrecognized by filename: /build/ubuntu-20.04-live-server-arm64.iso: N/A
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

This error indicates packer is trying to unpack the iso, even though I am specifying "iso" as `file_target_extension`. Looking at the code change in #14, I would expect to see the message "using raw image" instead of the message indicating that unpacking is being started

tannerjfco, 2020-07-08:
Looking a bit more into this it appears the ExtractAndCopyImage step is only included on `reuse` and `resize` image build methods. Setting to `reuse` produces the expected "using raw image" message, but the build still fails with a now less descriptive error

```
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm:local build boards/raspberry-pi-4/ubuntu.json
running /bin/packer build boards/raspberry-pi-4/ubuntu.json
arm: output will be in this color.

Warnings for build 'arm':

* a checksum type of 'none' was specified. Specyfying checksum is highly recommended
* a checksum type of 'none' was specified. Specyfying checksum is highly recommended

==> arm: Retrieving rootfs_archive
==> arm: Trying file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false
==> arm: Trying file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false
==> arm: file:///build/ubuntu-20.04-live-server-arm64.iso?archive=false => /build/ubuntu-20.04-live-server-arm64.iso
    arm: using raw image
    arm: searching for empty loop device (to map raspberry-pi-4.img)
    arm: mapping image raspberry-pi-4.img to /dev/loop24
    arm: mounting /dev/loop24p2 to /tmp/666436806
    arm: mounting /dev/loop24p1 to /tmp/666436806/boot
==> arm: exit status 32
==> arm: exit status 32
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

bradwood, 2020-08-09:
As a potential workaround, you could just use the Canonical-shipped `img` file, as per:

https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json#L5

bradwood, 2020-08-09:
Also, as per #41 you might need to change:

```
    "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-arm-static"

```  
to `qemu_binary_source_path": "/usr/bin/qemu-aarch64-static`

mkaczanowski, 2020-08-12:
@tannerjfco the issue is with the ubuntu image you're trying to use:
```
$ fdisk -l ubuntu-20.04.1-live-server-arm64.iso 
Disk ubuntu-20.04.1-live-server-arm64.iso: 880.11 MiB, 922859520 bytes, 1802460 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                                Boot   Start     End Sectors   Size Id Type
ubuntu-20.04.1-live-server-arm64.iso1           64 1801435 1801372 879.6M cd unknown
ubuntu-20.04.1-live-server-arm64.iso2      1801436 1802459    1024   512K ef EFI (FAT-12/16/32
```

First partion is not mountable, frankly speaking I have no idea what the partition type that is:
https://www.win.tue.nl/~aeb/partitions/partition_types-1.html

likely is somewhat compressed filesystem parsable by live loader (it's a live/installer image).

What you want is exactly what @bradwood did here:
> boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json

so feel free to use that recipe json file :)

closed 2020-08-12 by mkaczanowski