mkaczanowski/packer-plugin-builder-arm #45: jetson-nano: provision with apt update

BaptisteAmato, 2020-08-25:
Hi,

The creating of a jetson nano image does work well, thanks a lot for that :) 
I would like to run a "sudo apt-get update" command during the provisioning step, however I get "Failed to fetch" errors, so it does not seem to be able to connect.
Is it possible to use the internet of my local computer to do that?

Sorry if it's a noob question, I'm pretty new on Packer.

Best,

mkaczanowski, 2020-08-25:
```
root@ubuntu-bionic:~/packer-builder-arm# git diff
diff --git a/boards/jetson-nano/ubuntu.json b/boards/jetson-nano/ubuntu.json
index ec5a4c9..7ed3510 100644
--- a/boards/jetson-nano/ubuntu.json
+++ b/boards/jetson-nano/ubuntu.json
@@ -28,7 +28,9 @@
     {
       "type": "shell",
       "inline": [
-        "touch /tmp/test"
+        "touch /tmp/test",
+       "echo nameserver 8.8.8.8 > /etc/resolv.conf",
+        "apt-get update"
       ]
     }
   ],

```

```
root@ubuntu-bionic:~/packer-builder-arm# packer build boards/jetson-nano/ubuntu.json                                                                                                                                                                                                                                                                   
arm: output will be in this color.                                                                                                                                         
                                                                                                                                                                           
==> arm: Retrieving rootfs_archive                                                                                                                                         
==> arm: Trying https://developer.nvidia.com/jetson-nano-sd-card-image-r3221?archive=false                                                                                 
==> arm: Trying https://developer.nvidia.com/jetson-nano-sd-card-image-r3221?archive=false&checksum=md5%3A7b87e26d59c560ca18692a1ba282d842                                 
==> arm: https://developer.nvidia.com/jetson-nano-sd-card-image-r3221?archive=false&checksum=md5%3A7b87e26d59c560ca18692a1ba282d842 => /root/packer-builder-arm/packer_cache/fe16cb8795a3110a6cbcbf4b0812d132793d04d2.zip                                                                                                                              
    arm: unpacking /root/packer-builder-arm/packer_cache/fe16cb8795a3110a6cbcbf4b0812d132793d04d2.zip to jetson-nano.img                                                   
    arm: searching for empty loop device (to map jetson-nano.img)                                                                                                          
    arm: mapping image jetson-nano.img to /dev/loop0                                                                                                                       
    arm: mounting /dev/loop0p1 to /tmp/681727427                                                                                                                           
    arm: running extra setup                                                                                                                                               
    arm: mounting /dev with: [mount --bind /dev /tmp/681727427/dev]                                                                                                        
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/681727427/dev/pts]                                                                                           
    arm: mounting proc with: [mount -t proc proc /tmp/681727427/proc]                                                                                                      
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/681727427/proc/sys/fs/binfmt_misc]                                                              
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/681727427/sys]                                                                                                    
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/qemu-aarch64                                                                                                      
    arm: copying qemu binary from /usr/bin/qemu-aarch64-static to: /tmp/681727427/usr/bin/qemu-aarch64-static                                                              
    arm: running the provision hook                                                                                                                                        
==> arm: Provisioning with shell script: /tmp/packer-shell637176744                                                                                                        
    arm: Get:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease                                                                                                          
    arm: Ign:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease                                                                                                          
    arm: Get:2 file:/var/visionworks-repo  InRelease                                                                                                                       
    arm: Ign:2 file:/var/visionworks-repo  InRelease                                                                                                                       
    arm: Get:3 file:/var/visionworks-sfm-repo  InRelease                                                                                                                                                                                                                                                                                               
[... cut ...]
    arm: Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]                                                                                    
    arm: Get:16 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]                    
    arm: Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]                                                                                   
    arm: Ign:18 http://international.download.nvidia.com/jetson/repos/<SOC> r32 InRelease       
....
...
...
```

closed 2020-08-25 by mkaczanowski

BaptisteAmato, 2020-08-26:
Seems to solve the internet connection! However I get an error when it goes to the repo.download.nvidia:

`
    arm: Get:75 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 64x64 Icons [109 kB]
    arm: Get:76 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse arm64 Packages [1,848 B]
    arm: Get:77 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse Translation-en [2,880 B]
    arm: Get:78 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse DEP-11 48x48 Icons [29 B]
    arm: Get:79 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse DEP-11 64x64 Icons [2,638 B]
    arm: Get:80 https://repo.download.nvidia.com/jetson/common r32.4/main arm64 Packages [13.5 kB]
    arm: Reading package lists...
==> arm: E: The repository 'https://repo.download.nvidia.com/jetson/<SOC> r32.4 Release' does not have a Release file.
==> arm: error while running provision hook: Script exited with non-zero exit status: 100.Allowed exit codes are: [0]
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' errored: build was halted
`

mkaczanowski, 2020-08-26:
that's the problem of the image itself, not the packer plugin

BaptisteAmato, 2020-08-26:
Got it, thanks!

BaptisteAmato, 2020-08-26:
Actually I'm not sure it comes from the image. I removed all provisions, did create my image, put it on a SD card and launched the system. After creating the first user, I was able to successfully do the "sudo apt update" command.
Does it mean I need to do the user configuration inside the provisioning before updating?