mkaczanowski/packer-plugin-builder-arm #47: ==> arm: Unknown host QEMU_IFLA type: 50

BaptisteAmato, 2020-08-27:
Hi,

I get several erros like this one during provisioning:
`
==> arm: Unknown host QEMU_IFLA type: XXX
`

Any idea where it comes from?

mkaczanowski, 2020-08-27:
those are the qemu binfmt messages

BaptisteAmato, 2020-08-27:
Does it mean everything related to qemu binfmt is about the image I'm building? I'm having a hard time understanding where the commands come from, the host or the newly created image...

mkaczanowski, 2020-08-27:
packer-builder-arm chroot's to the image, so it's always a host (in a sense, there is no virtualization). binfmt + qemu is just a way to emulate different CPU architecture.

The problem is somewhere with your qemu or smth system related (not packer-builder-arm related) thus closing

closed 2020-08-27 by mkaczanowski