mkaczanowski/packer-plugin-builder-arm #49: Unable to mount (exit status 32)

ferrarimarco, 2020-08-27:
Hi! I'm trying to build OS images with this packer plugin for different boards (BeagleBone Black, Raspberry Pi 4).

Regardless of the template I use (tried with the ones that are in this repository, without modification), I get this error at the mounting step:

```shell
==> arm: exit status 32
```

Host system:

- Packer version: 1.6.1
- packer-builder-arm version: latest master.
- Ubuntu 20.04.1

Command I'm running: `packer build boards/beaglebone-black/debian.json`. The source of the template the [vanilla one that's in this repository](https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/beaglebone-black/debian.json).

If I try to mount the image without specifying the offset, I get the same error. If I set the "offset" option, I'm able to mount without any issue:

```shell
$ sudo mount -o "discard" "$dev" "$dst"                                                                                                                                                                                                                                                   
mount: /mnt/loop0: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.

$ sudo mount -o discard,offset=4194304 "$dev" "$dst"

$ ls $dst
bbb-uEnv.txt  bin  boot  dev  etc  home  ID.txt  lib  lost+found  media  mnt  nfs-uEnv.txt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
```

`parted` output (from which I calculated the offset):

```shell
Model:  (file)
Disk bone-debian-9.9-iot-armhf-2019-08-03-4gb.img: 7372800s
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start  End       Size      Type     File system  Flags
 1      8192s  7372799s  7364608s  primary  ext4         boot
```

Any help would be greatly appreciated :) Thanks!

Full Packer output:

```shell
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=sha256%3Aaadc64a1d069c842e56a4289fe1a6b4b5a0af4efcf95bcce78eb2a80fe5270f4

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 81.70 KiB / 676.26 MiB    0.01% 28m18s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 455.01 KiB / 676.26 MiB    0.07% 10m08s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 2.80 MiB / 676.26 MiB    0.41% 02m24s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 9.98 MiB / 676.26 MiB    1.48% 00m53s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 17.13 MiB / 676.26 MiB    2.53% 00m38s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 23.95 MiB / 676.26 MiB    3.54% 00m32s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 31.65 MiB / 676.26 MiB    4.68% 00m28s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 38.98 MiB / 676.26 MiB    5.76% 00m26s

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 46.29 MiB / 676.26 MiB    6.84% 00m24s

[... 184 lines cut ...]
ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 653.89 MiB / 676.26 MiB   96.69%

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 661.39 MiB / 676.26 MiB   97.80%

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 667.87 MiB / 676.26 MiB   98.76%

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 675.36 MiB / 676.26 MiB   99.87%

ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz 676.26 MiB / 676.26 MiB  100.00% 21s
==> arm: http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=sha256%3Aaadc64a1d069c842e56a4289fe1a6b4b5a0af4efcf95bcce78eb2a80fe5270f4 => /workspace/packer_cache/c139ba11167a4880e98a4525349709cd49b02855.xz
    arm: unpacking /workspace/packer_cache/c139ba11167a4880e98a4525349709cd49b02855.xz to ubuntu-20.04.img
    arm: unpacking with custom comand: [xz --decompress /tmp/image292925921/c139ba11167a4880e98a4525349709cd49b02855.xz]
    arm: searching for empty loop device (to map ubuntu-20.04.img)
    arm: mapping image ubuntu-20.04.img to /dev/loop4
    arm: mounting /dev/loop4p2 to /tmp/873780172
==> arm: exit status 32
==> arm: exit status 32
==> arm: exit status 32
Build 'arm' errored: build was halted

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

mkaczanowski, 2020-08-27:
Works for me:
```
$ sudo packer build boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=sha256%3Aaadc64a1d069c842e56a4289fe1a6b4b5a0af4efcf95bcce78eb2a80fe5270f4
==> arm: http://cdimage.ubuntu.com/releases/20.04.1/release/ubuntu-20.04.1-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=sha256%3Aaadc64a1d069c842e56a4289fe1a6b4b5a0af4efcf95bcce78eb2a80fe5270f4 => /home/admin/Projects/packer-builder-arm/packer_cache/c139ba11167a4880e98a4525349709cd49b02855.xz
    arm: unpacking /home/admin/Projects/packer-builder-arm/packer_cache/c139ba11167a4880e98a4525349709cd49b02855.xz to ubuntu-20.04.img
    arm: unpacking with custom comand: [xz --decompress /tmp/image332588658/c139ba11167a4880e98a4525349709cd49b02855.xz]
    arm: searching for empty loop device (to map ubuntu-20.04.img)
    arm: mapping image ubuntu-20.04.img to /dev/loop2
    arm: mounting /dev/loop2p2 to /tmp/340243753
    arm: mount -o discard /dev/loop2p2 /tmp/340243753
    arm: mounting /dev/loop2p1 to /tmp/340243753/boot/firmware
    arm: mount -o discard /dev/loop2p1 /tmp/340243753/boot/firmware
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/340243753/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/340243753/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/340243753/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/340243753/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/340243753/sys]
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/aarch64
    arm: copying qemu binary from /usr/bin/qemu-aarch64-static to: /tmp/340243753/usr/bin/qemu-aarch64-static
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell373148472
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' finished.

==> Builds finished. The artifacts of successful builds are:
--> arm: ubuntu-20.04.img

```

```
[admin@laptop packer-builder-arm]$ sudo packer build boards/beaglebone-black/debian.json                                                                                    
arm: output will be in this color.   
                                                                                     
==> arm: Retrieving rootfs_archive   
==> arm: Trying https://debian.beagleboard.org/images/bone-debian-9.9-iot-armhf-2019-08-03-4gb.img.xz?archive=false
==> arm: Trying https://debian.beagleboard.org/images/bone-debian-9.9-iot-armhf-2019-08-03-4gb.img.xz?archive=false&checksum=md5%3A363c8bbc87759705629f7550f2e54de2
bone-debian-9.9-iot-armhf-2019-08-03-4gb.img.xz 727.69 MiB / 727.69 MiB [===============================================================================================================================================================================================================================================================] 100.00% 8m32s
==> arm: https://debian.beagleboard.org/images/bone-debian-9.9-iot-armhf-2019-08-03-4gb.img.xz?archive=false&checksum=md5%3A363c8bbc87759705629f7550f2e54de2 => /home/admin/Projects/packer-builder-arm/packer_cache/64f1a9dddf78dd75048766efccc3c1fa89f6cce4.xz
    arm: unpacking /home/admin/Projects/packer-builder-arm/packer_cache/64f1a9dddf78dd75048766efccc3c1fa89f6cce4.xz to beaglebone-black.img
    arm: unpacking with custom comand: [xz -d /tmp/image256875106/64f1a9dddf78dd75048766efccc3c1fa89f6cce4.xz]
    arm: searching for empty loop device (to map beaglebone-black.img)                                                                                                     
    arm: mapping image beaglebone-black.img to /dev/loop2
    arm: mounting /dev/loop2p1 to /tmp/213535321
    arm: mount -o discard /dev/loop2p1 /tmp/213535321
    arm: running extra setup      
    arm: mounting /dev with: [mount --bind /dev /tmp/213535321/dev]                                                                                                        
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/213535321/dev/pts]                                                                                                                                                                                                                                                                       
    arm: mounting proc with: [mount -t proc proc /tmp/213535321/proc]                                                                                                                                                                                                                                                                                  
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/213535321/proc/sys/fs/binfmt_misc]                          
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/213535321/sys]                                                                                                    
    arm: binfmt setup found at: /proc/sys/fs/binfmt_misc/arm      
    arm: copying qemu binary from /usr/bin/qemu-arm-static to: /tmp/213535321/usr/bin/qemu-arm-static
    arm: running the provision hook             
==> arm: Provisioning with shell script: /tmp/packer-shell558248344
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' finished.                                              
                                                                                     
==> Builds finished. The artifacts of successful builds are:       
--> arm: beaglebone-black.img   
````

did you run with root?

mkaczanowski, 2020-08-27:
(also works in the github pipeline)

mkaczanowski, 2020-08-27:
closing since it's not plugin related, it's rather your environment.

You might want to test it out in the VM to see the problem is somewhere in your local setup:
https://gist.github.com/mkaczanowski/c77b03323d9f376e75c9e91063213b36

closed 2020-08-27 by mkaczanowski

ferrarimarco, 2020-08-28:
OK thanks!