mkaczanowski/packer-plugin-builder-arm #52: Binary releases?

eduncan911, 2020-08-29:
Hello!

I see you are already accustomed to github actions.  Why not add some actions to build and attach the binaries for macOS and Linux?

That way, we don't have to tell users to install Golang to compile it.  They can just wget the latest release.

I mention this because I am writing a few blog posts for less-Linux inclined people and would rather they just run a `wget`.

mkaczanowski, 2020-08-30:
> That way, we don't have to tell users to install Golang to compile it. They can just wget the latest release.

Taken there is a few other deps like (kpartx, parted, compatible packer binary) the Docker is a better choice. While I have nothing against pushing a new binary with the release, I don't see how is that better than simple `docker run ....`

closed 2020-08-30 by mkaczanowski

eduncan911, 2020-08-30:
>  I don't see how is that better than simple docker run ....

That's easy.  GoLang nor Docker are required to stay on the user's machine once it is compiled.  Only the built binary is for ongoing runs.

mkaczanowski, 2020-08-30:
> Only the built binary is needed for ongoing runs.
As explained earlier that statement is not true. You need some deps to be set up, either in your system or the docker image. 
Taken that most of the people don't realize they're missing a dependency (ie. parted) they'd rather blame the plugin rather their environment... So, docker definitely is a better choice for 90% of the people.

> Why would you want users to install full dev environments ...
I don't, so use docker and all problems go away, as simple as that.

----

Please take note, I am fine with both solutions, but for the reasons mentioned in this thread I won't invest time in setting up that pipeline/action, however, PR's are most welcome

eduncan911, 2020-08-30:
Ah, there are other deps that the plugin needs.

faddat, 2020-09-04:
There shall be a PR.  Oner of my colleagues is suspicious of binary docker distributions, so.... PR time.  

PR shall be:

* Bash install script which assumes it's running on Ubuntu 20.04
-- but probably not binaries.

eduncan911, 2020-09-04:
Sounds like your colleague wouldn't like aptitude/apt-get, pacman, yum, snapd, AppImage, flatpak, etc either as they are all binary download as well 

There's nothing to be suspecious about if authors include a SHA hash for every binary download. Not only does a hash guarantee the integrity of the download, but also allows one to run the CICD/make from source to verify the binary isn't compromised.

If authors don't release the source for a binary, then suspecious is warranted - like the Discord AppImage, there's no way anyone should be running that thing with it's missing source code because who knows what's inside that AppImage.

But if deps are required on a system to run the plugin, that's a different discussion.  And this is where releasing a Docker build environment makes better sense - to keep the deps off of the host machine, and to better control the deps in an isolated environment (within that Docker build env).

faddat, 2020-09-09:
my colleague is suspicious of everything, including all those tools you named.