mkaczanowski/packer-plugin-builder-arm #57: Network issues with aarch64 on rpi board config

tweidinger, 2020-10-01:
Hi, first of all thanks for this plugin as it saved me a lot of hours before.

Currently I am setting up a raspberry pi 4 and got confronted with multiple issues.
Reference issues are in this repo: #16 in the solo-io repo: https://github.com/solo-io/packer-builder-arm-image/issues/58  https://github.com/solo-io/packer-builder-arm-image/pull/50 I found there was a workaround implemented 
but this seems to be not the case in this repo.

The problem is affecting the discussed issue of using the provisioning to update 
the system with network access.
I used https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-3/archlinuxarm.json as a reference.

```
==> arm: error: failed retrieving file 'core.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm: error: failed to update core (invalid url for server)
==> arm: error: failed retrieving file 'extra.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm: error: failed to update extra (invalid url for server)
==> arm: error: failed retrieving file 'community.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm: error: failed to update community (invalid url for server)
==> arm: error: failed retrieving file 'alarm.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm: error: failed to update alarm (invalid url for server)
==> arm: error: failed retrieving file 'aur.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm: error: failed to update aur (invalid url for server)
==> arm: error: failed to synchronize all databases
==> arm: error while running provision hook: Script exited with non-zero exit status: 1.Allowed exit codes are: [0]
==> arm: optional `fuser -k` failed with exit status 1:
Build 'arm' errored after 25 seconds 349 milliseconds: build was halted
```

I printed the content of _/etc/resolv.conf_ while provisioning and got the `nameserver 8.8.8.8` result.
When trying to ping any domain I get `arm: /usr/bin/ping: google.com: Temporary failure in name resolution`.

I tried to implement workarounds with using the strategy from the solo-io repo and did manual mounts, without mounting the resolv.conf and instead mounting /run/systemd as additional mount. This gave me the same error.
Host sytem: Archlinux 5.7 kernel inside kvm.
Internet access on host system is working.

Any ideas to proceed?

mkaczanowski, 2020-10-02:
I can repro this locally, so smth had to change...

I am wondering if this is qemu update, libc or kernel. Interestingly it works fine for beaglebone:
```
sudo packer build boards/beaglebone-black/archlinuxarm.json
```

I'll have to do some troubleshooting

mkaczanowski, 2020-10-16:
The issue is gone with latest Archlinux update.

https://github.com/mkaczanowski/packer-builder-arm/commit/ca89d6308bc6c6af04529b755c87ed92dd66951c

I've added the Vagrantfile to help troubleshoot further (often platform-specific) issues: https://github.com/mkaczanowski/packer-builder-arm/commit/ca89d6308bc6c6af04529b755c87ed92dd66951c

closed 2020-10-16 by mkaczanowski