mkaczanowski/packer-plugin-builder-arm #61: Failed to find binfmt_misc for /usr/bin/qemu-aarch64-static under /proc/sys/fs/binfmt_misc

fguiet, 2020-11-29:
Hi,

I try to build a raspberry image using your Docker image...(packer 1.6.5) but it fails : 

`==> arm: Failed to find binfmt_misc for /usr/bin/qemu-aarch64-static under /proc/sys/fs/binfmt_misc`

I am using `2020-08-20-raspios-buster-arm64-lite.zip` base image.

Here is my json file : [raspios-lite-arm64.zip](https://github.com/mkaczanowski/packer-builder-arm/files/5612780/raspios-lite-arm64.zip)

Any idea?

Fred

Log below :

```bash
[root@localhost packer-builder-arm]# docker run --rm --privileged -v /dev:/dev -v /applications/packer-builder-arm:/build  -v /applications/raspios:/applications/raspios -v /applications/packer-builder-arm/img:/img packer-builder-arm build /img/raspbian_test.json
running /bin/packer build /img/raspbian_test.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying /applications/raspios/2020-08-20-raspios-buster-arm64-lite.zip?archive=false
==> arm: Trying /applications/raspios/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=sha256%3A0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed
==> arm: /applications/raspios/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=sha256%3A0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed => /applications/raspios/2020-08-20-raspios-buster-arm64-lite.zip
    arm: unpacking /applications/raspios/2020-08-20-raspios-buster-arm64-lite.zip to raspios-arm64.img
    arm: searching for empty loop device (to map raspios-arm64.img)
    arm: mapping image raspios-arm64.img to /dev/loop0
    arm: mounting /dev/loop0p2 to /tmp/918012778
    arm: mounting /dev/loop0p1 to /tmp/918012778/boot
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/918012778/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/918012778/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/918012778/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/918012778/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/918012778/sys]
==> arm: Failed to find binfmt_misc for /usr/bin/qemu-aarch64-static under /proc/sys/fs/binfmt_misc
==> arm: remove /tmp/918012778/usr/bin/qemu-aarch64-static: no such file or directory
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' errored after 16 seconds 322 milliseconds: build was halted

==> Wait completed after 16 seconds 322 milliseconds

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

mkaczanowski, 2020-11-29:
It seems you didn't setup binftm feature (https://en.wikipedia.org/wiki/Binfmt_misc). See: https://wiki.archlinux.org/index.php/QEMU (or other relevant to your distro/OS documentation).

closed 2020-11-29 by mkaczanowski

fguiet, 2020-11-29:
I need to set up binftm even I am using your Docker image? 
Have you got any documentation about this?
Thank
Fred

mkaczanowski, 2020-11-29:
Please take a breath and read the wiki :) It's a kernel feature, so yeah you need to have it enabled before you run docker

> is a capability of the Linux kernel which allows arbitrary executable file formats to be recognized and passed to certain user space applications, such as emulators and virtual machines
> ... The executable formats are registered through the special purpose file system binfmt_misc file-system interface (usually mounted under part of /proc).

fguiet, 2020-11-29:
Sorry, thank you mate !

fguiet, 2020-11-30:
Just to share what I've done to others  : switch from CentOS 7 image (find binfmt-support package is quite impossible...) to Debian 10.6 netinst amd64 image.
I installed `binfmt-support` package...now it's working like a charm 
Fred