mkaczanowski/packer-plugin-builder-arm #68: RPi 4 Ubuntu Provisioners don't have networking

MikeTurpinSDG, 2021-01-19:
Steps to Reproduce:

1. Modify provisioners second in boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json to:
```
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "apt update",
        "apt upgrade -y"
      ]
    }
  ],
```

2. Commit and push changes to new remote repo.
3.  Modify line 20 in vagrant file to point to new remote repo.
4. Modify line 25 in vagrant file to:
packer build boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json
5. run vagrant up

In the terminal, once the provisioner runs, you will see the following errors:
```
  default:     arm: running the provision hook
    default: ==> arm: Provisioning with shell script: /tmp/packer-shell796640409
    default: ==> arm:
    default: ==> arm: WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
    default: ==> arm:
    default:     arm: Err:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
    default:     arm:   Temporary failure resolving 'ports.ubuntu.com'
    default:     arm: Err:2 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
    default:     arm:   Temporary failure resolving 'ports.ubuntu.com'
    default:     arm: Err:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
    default:     arm:   Temporary failure resolving 'ports.ubuntu.com'
```
From what I can tell it appears that the QEMU networking hasn't been setup or initialized.    But not sure if that is expected or not. I created the error on both Ubuntu 20.04 running packer-builder-arm directly and in the vagrant image. I attempted to setup a bridge have not been successful.  Please let me know if there is any additional information I can provide or if this this not supported functionality then I can continue working to determine how to establish network connectivity to the packer provisioner.

mkaczanowski, 2021-01-19:
Have you tried to update the dns resolver?
ie. https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/archlinuxarm.json#L42

MikeTurpinSDG, 2021-01-20:
Yep that was it, figured there was a good chance it was something simple like that. Thanks for your quick response!

For anyone stumbling on this issue.  I added 

`{
      "type": "shell",
      "inline": [
        "rm /etc/resolv.conf",
        "echo 'nameserver 8.8.8.8' > /etc/resolv.conf"
      ]
    },`
    
    The issue when I first tried it was that /etc/resolv.conf was symlinked  to /run/systemd/resolve/resolv.conf and for some reason couldn't update the linked file.  Removing the link allowed me to get networking to function correctly.

closed 2021-01-20 by MikeTurpinSDG

Snarik, 2021-03-06:
So glad I found this-- Had this exact problem. Would it be worth while to submit a PR for an update to the packer configs?