mkaczanowski/packer-plugin-builder-arm #76: Building arm64 on arm64 RPi: "Failed to find binfmt_misc for /usr/bin/qemu-arm-static under /proc/sys/fs/binfmt_misc"

jgspratt, 2021-03-24:
I downgraded to Pakcer v1.6.6 and ran into a different issue:

```
# hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: bc884c3eac3149239aa76fb02fd96f92
           Boot ID: 3189d8eb53774efda0c9f8cc084333e4
  Operating System: Debian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.17-v8+
      Architecture: arm64
```

```
# ./build.sh
++ cd .
++ pwd
+ THIS_DIR=/root/git/imager
+ export PACKER_PLUGIN_PATH=/root/git/imager/plugins
+ PACKER_PLUGIN_PATH=/root/git/imager/plugins
+ export PACKER_LOG=1
+ PACKER_LOG=1
+ packer build /root/git/imager/img/arm64/packer.json
2021/03/24 02:55:54 [INFO] Packer version: 1.6.6 [go1.15.6 linux arm64]
2021/03/24 02:55:54 [DEBUG] Discovered plugin: arm = /root/git/imager/plugins/packer-builder-arm
2021/03/24 02:55:54 using external builders [arm]
2021/03/24 02:55:54 Checking 'PACKER_CONFIG' for a config file path
2021/03/24 02:55:54 'PACKER_CONFIG' not set; checking the default config file path
2021/03/24 02:55:54 Attempting to open config file: /root/.packerconfig
2021/03/24 02:55:54 [WARN] Config file doesn't exist: /root/.packerconfig
2021/03/24 02:55:54 Setting cache directory: /root/git/imager/packer_cache
2021/03/24 02:55:54 Creating plugin client for path: /root/git/imager/plugins/packer-builder-arm
2021/03/24 02:55:54 Starting plugin: /root/git/imager/plugins/packer-builder-arm []string{"/root/git/imager/plugins/packer-builder-arm"}
2021/03/24 02:55:54 Waiting for RPC address for: /root/git/imager/plugins/packer-builder-arm
2021/03/24 02:55:54 packer-builder-arm plugin: 2021/03/24 02:55:54 Plugin address: unix /tmp/packer-plugin165252807
2021/03/24 02:55:54 packer-builder-arm plugin: 2021/03/24 02:55:54 Waiting for connection...
2021/03/24 02:55:54 Received unix RPC address for /root/git/imager/plugins/packer-builder-arm: addr is /tmp/packer-plugin165252807
2021/03/24 02:55:54 packer-builder-arm plugin: 2021/03/24 02:55:54 Serving a plugin connection...
2021/03/24 02:55:54 Creating plugin client for path: /usr/local/bin/packer
2021/03/24 02:55:54 Starting plugin: /usr/local/bin/packer []string{"/usr/local/bin/packer", "plugin", "packer-provisioner-shell"}
2021/03/24 02:55:54 Waiting for RPC address for: /usr/local/bin/packer
2021/03/24 02:55:54 packer-provisioner-shell plugin: [INFO] Packer version: 1.6.6 [go1.15.6 linux arm64]
2021/03/24 02:55:54 packer-provisioner-shell plugin: Checking 'PACKER_CONFIG' for a config file path
2021/03/24 02:55:54 packer-provisioner-shell plugin: 'PACKER_CONFIG' not set; checking the default config file path
2021/03/24 02:55:54 packer-provisioner-shell plugin: Attempting to open config file: /root/.packerconfig
2021/03/24 02:55:54 packer-provisioner-shell plugin: [WARN] Config file doesn't exist: /root/.packerconfig
2021/03/24 02:55:54 packer-provisioner-shell plugin: Setting cache directory: /root/git/imager/packer_cache
2021/03/24 02:55:54 packer-provisioner-shell plugin: args: []string{"packer-provisioner-shell"}
2021/03/24 02:55:54 packer-provisioner-shell plugin: Plugin address: unix /tmp/packer-plugin985452285
2021/03/24 02:55:54 Received unix RPC address for /usr/local/bin/packer: addr is /tmp/packer-plugin985452285
2021/03/24 02:55:54 packer-provisioner-shell plugin: Waiting for connection...
2021/03/24 02:55:54 packer-provisioner-shell plugin: Serving a plugin connection...
2021/03/24 02:55:54 Preparing build: arm
2021/03/24 02:55:54 Build debug mode: false
2021/03/24 02:55:54 Force build: false
2021/03/24 02:55:54 On error:
2021/03/24 02:55:54 Waiting on builds to complete...
2021/03/24 02:55:54 Starting build run: arm
arm: output will be in this color.

2021/03/24 02:55:54 Running builder: arm
2021/03/24 02:55:54 [INFO] (telemetry) Starting builder arm
==> arm: Retrieving rootfs_archive
2021/03/24 02:55:54 packer-builder-arm plugin: 2021/03/24 02:55:54 Acquiring lock for: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=sha256%3A0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed (/root/git/imager/packer_cache/0a4bce9913c2935e03ed1c1b6914cc98e96e8bee.zip.lock)
==> arm: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false
==> arm: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=sha256%3A0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed
==> arm: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=sha256%3A0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed => /root/git/imager/packer_cache/0a4bce9913c2935e03ed1c1b6914cc98e96e8bee.zip
2021/03/24 02:56:07 packer-builder-arm plugin: 2021/03/24 02:56:07 Leaving retrieve loop for rootfs_archive
    arm: unpacking /root/git/imager/packer_cache/0a4bce9913c2935e03ed1c1b6914cc98e96e8bee.zip to raspberry-pi-arm64-out.img
    arm: searching for empty loop device (to map raspberry-pi-arm64-out.img)
    arm: mapping image raspberry-pi-arm64-out.img to /dev/loop0
    arm: mounting /dev/loop0p2 to /tmp/518139580
    arm: mounting /dev/loop0p1 to /tmp/518139580/boot
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/518139580/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/518139580/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/518139580/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/518139580/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/518139580/sys]
==> arm: Failed to find binfmt_misc for /usr/bin/qemu-arm-static under /proc/sys/fs/binfmt_misc
==> arm: remove /tmp/518139580/usr/bin/qemu-arm-static: no such file or directory
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
2021/03/24 02:58:11 [INFO] (telemetry) ending arm
==> Wait completed after 2 minutes 16 seconds
2021/03/24 02:58:11 machine readable: error-count []string{"1"}
==> Some builds didn't complete successfully and had errors:
2021/03/24 02:58:11 machine readable: arm,error []string{"build was halted"}
==> Builds finished but no artifacts were created.
Build 'arm' errored after 2 minutes 16 seconds: build was halted

==> Wait completed after 2 minutes 16 seconds

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
2021/03/24 02:58:11 [INFO] (telemetry) Finalizing.
2021/03/24 02:58:12 waiting for all plugin processes to complete...
2021/03/24 02:58:12 /root/git/imager/plugins/packer-builder-arm: plugin process exited
2021/03/24 02:58:12 /usr/local/bin/packer: plugin process exited
```

I did end up with this file as output:

```
-rw-r--r-- 1 root root 1.9G 2021-03-24 02:58 raspberry-pi-arm64-out.img
```

I have not tested it to see if it is viable.

I did find this code block in `imager/plugins/src/packer-builder-arm/config/image_config.go`:

```
	if len(c.ImageChrootMounts) == 0 {
		c.ImageChrootMounts = []ChrootMount{
			{MountType: "proc", SourcePath: "proc", DestinationPath: "/proc"},
			{MountType: "sysfs", SourcePath: "sysfs", DestinationPath: "/sys"},
			{MountType: "bind", SourcePath: "/dev", DestinationPath: "/dev"},
			{MountType: "devpts", SourcePath: "/devpts", DestinationPath: "/dev/pts"},
			{MountType: "binfmt_misc", SourcePath: "binfmt_misc", DestinationPath: "/proc/sys/fs/binfmt_misc"},
		}
	}
```

I will try overriding it.

jgspratt, 2021-03-24:
Trying inside docker:

```
# docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm build boards/raspberry-pi-3/raspios-lite-arm64.json
Unable to find image 'mkaczanowski/packer-builder-arm:latest' locally
latest: Pulling from mkaczanowski/packer-builder-arm
5d3b2c2d21bb: Pull complete
3fc2062ea667: Pull complete
75adf526d75b: Pull complete
fa9b69dc9297: Pull complete
19543ba00220: Pull complete
7f72a9ad5699: Pull complete
9dbce454229e: Pull complete
7045fbe28d35: Pull complete
1ec39da9c97d: Pull complete
Digest: sha256:9a266a942ce958b8c7eaa241c587b9355609cab987e5811e54c2e3f996c73f57
Status: Downloaded newer image for mkaczanowski/packer-builder-arm:latest
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
running /bin/packer build boards/raspberry-pi-3/raspios-lite-arm64.json
/entrypoint.sh: line 27: /bin/packer: No such file or directory
```

dbast, 2021-03-24:
The Docker Image is a linux/amd64 ... that is not expected to run on arm64 rpi.

Also building an arm64 image while being on arm64, does not require all the emulation stuff with qemu.

jgspratt, 2021-03-24:
That makes sense.

What changes to the provider .json files would you recommend in order to build a custom raspberry pi image?

jgspratt, 2021-03-24:
For the record, I tried this, but it exhibited the error I mentioned above:

```
{
  "variables": {},
  "builders": [{
    "type": "arm",
    "file_urls" : ["https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip"],
    "file_checksum_url": "https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip.sha256",
    "file_checksum_type": "sha256",
    "file_target_extension": "zip",
    "image_build_method": "reuse",
    "image_path": "raspios-arm64.img",
    "image_size": "2G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "8192",
        "filesystem": "vfat",
        "size": "256M",
        "mountpoint": "/boot"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "532480",
        "filesystem": "ext4",
        "size": "0",
        "mountpoint": "/"
      }
    ],
    "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
    "qemu_binary_source_path": "/usr/bin/qemu-aarch64-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-aarch64-static"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "touch /tmp/test"
      ]
    }
  ]
}
```

Can I just omit mentions of `qemu` in the json?

jgspratt, 2021-04-07:
Update: I never got this to work and instead used the major completing project.

I'm still open to coming back and using this one if my question above were answered.

dbast, 2021-04-09:
@jgspratt This should work for you now on Linux aarch64/arm64 https://github.com/mkaczanowski/packer-builder-arm/pull/78

You need to get my master branch and locally build the docker image instead of getting it from Dockerhub:
* git clone https://github.com/dbast/packer-builder-arm && cd packer-builder-arm
* docker build -t packer-builder-arm -f docker/Dockerfile .
* docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi-3/raspios-lite-arm64.json

dbast, 2021-04-12:
@jgspratt Pushed a test version to Dockerhub ... so you can directly test it via:
`docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build dbast/packer-builder-arm build boards/raspberry-pi-3/raspios-lite-arm64.json`... it is a multiarch image that runs on linux/amd64 and linux/aarch64.

closed 2021-04-12 by mkaczanowski

dvogel, 2021-06-19:
@dbast running those commands from commit 9b03c4833ee9a71d91d1dd0461d0f3849cb76c3a (current tip of the `master` branch), I get an error:

```
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi-3/raspios-lite-arm64.json
running /bin/packer build boards/raspberry-pi-3/raspios-lite-arm64.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false
==> arm: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=file%3Ahttps%3A%2F%2Fdownloads.raspberrypi.org%2Fraspios_lite_arm64%2Fimages%2Fraspios_lite_arm64-2020-08-24%2F2020-08-20-raspios-buster-arm64-lite.zip.sha256
==> arm: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip?archive=false&checksum=file%3Ahttps%3A%2F%2Fdownloads.raspberrypi.org%2Fraspios_lite_arm64%2Fimages%2Fraspios_lite_arm64-2020-08-24%2F2020-08-20-raspios-buster-arm64-lite.zip.sha256 => /build/packer_cache/ef982058ce04ced0a52f6545379b1b40ca4f371f.zip
    arm: unpacking /build/packer_cache/ef982058ce04ced0a52f6545379b1b40ca4f371f.zip to raspios-arm64.img
    arm: searching for empty loop device (to map raspios-arm64.img)
    arm: mapping image raspios-arm64.img to /dev/loop0
    arm: mounting /dev/loop0p2 to /tmp/282680203
    arm: mounting /dev/loop0p1 to /tmp/282680203/boot
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/282680203/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/282680203/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/282680203/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/282680203/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/282680203/sys]
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell754194613
==> arm: chroot: failed to run command '/bin/sh': No such file or directory
==> arm: error while running provision hook: Script exited with non-zero exit status: 127.Allowed exit codes are: [0]
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' errored after 3 minutes 19 seconds: build was halted

==> Wait completed after 3 minutes 19 seconds

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

dbast, 2021-06-19:
@dvogel works here... 

Does using the current image from dockerhub work?:

```docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm build boards/raspberry-pi-3/raspios-lite-arm64.json```

What is the host OS (linux, osx) and arch (x86-64, arm64/aarch64)? What is the used docker version?