mkaczanowski/packer-plugin-builder-arm #83: rpi4 ubuntu sample not provisionning

omartin-ndlp, 2021-04-17:
Hey guys,

I modified the ubuntu_server_20.04_arm64.json sample in the raspberry-4 directory. Just basic stuff:  provisioning with apt-get update and apt-get install, but it doesn't work. I'm getting "Temporary failure in name resolution" errors. 

It's not an issue with networking because ping'ing IP addresses works. Afaik it's a matter of host name resolution:  it cannot resolve host names in the chroot-ed file system. The file resolv.conf file points to a file that should be in the /run directory but is not. Here's a terminal dump that illustrates the issue:

```bash
olivier@tyrion:/tmp/105512316$ ls -l
total 68
lrwxrwxrwx   1 root root     7 mars  31 09:45 bin -> usr/bin
drwxr-xr-x   3 root root  4096 mars  31 09:50 boot
drwxr-xr-x  20 root root  4500 avril 17 08:31 dev
drwxr-xr-x  96 root root  4096 mars  31 09:52 etc
drwxr-xr-x   2 root root  4096 mars   2 23:33 home
lrwxrwxrwx   1 root root     7 mars  31 09:45 lib -> usr/lib
drwx------   2 root root 16384 mars  31 09:53 lost+found
drwxr-xr-x   2 root root  4096 mars  31 09:45 media
drwxr-xr-x   2 root root  4096 mars  31 09:45 mnt
drwxr-xr-x   2 root root  4096 mars  31 09:45 opt
dr-xr-xr-x 304 root root     0 avril 17 08:31 proc
drwx------   2 root root  4096 mars  31 09:50 root
drwxr-xr-x   3 root root  4096 mars  31 09:52 run
lrwxrwxrwx   1 root root     8 mars  31 09:45 sbin -> usr/sbin
drwxr-xr-x   6 root root  4096 mars  31 09:52 snap
drwxr-xr-x   2 root root  4096 mars  31 09:45 srv
dr-xr-xr-x  13 root root     0 avril 16 22:53 sys
drwxrwxrwt   2 root root  4096 mars  31 09:50 tmp
drwxr-xr-x  12 root root  4096 mars  31 09:48 usr
drwxr-xr-x  13 root root  4096 mars  31 09:50 var
olivier@tyrion:/tmp/105512316$ ls -l etc/resolv.conf
lrwxrwxrwx 1 root root 39 mars  31 09:46 etc/resolv.conf -> ../run/systemd/resolve/stub-resolv.conf
olivier@tyrion:/tmp/105512316$ ls -l run
total 4
drwxr-xr-x 2 root root 4096 mars  31 09:52 mount
olivier@tyrion:/tmp/105512316$
```
The sample works as-is (I mean that I tried the image produced by the sample on a RPI4), so I guess something is done at boot time that packer-builder-arm doesn't simulate. Also the raspbian sample works when modified for provisioning with apt-get, and I'm on Ubuntu 20.10 with the PPA for qemu 5.2.

Thanks, and apologies if it's not a packer-builder-arm issue, as I'm way out of my comfort zone here.

dbast, 2021-04-17:
maybe this works:
* backup symlink: mv etc/resolv.conf etc/resolv.conf.bak
* configure via real file: echo "8.8.8.8" > etc/resolv.conf
* ... do all the apt-get install
* restore symlink: rm etc/resolv.conf && mv etc/resolv.conf.bak etc/resolv.conf

omartin-ndlp, 2021-04-17:
Yep, that's what I tried (copied from the archlinux sample) and it works. Thanks for the suggestion, I'll submit a fix when I get around to it. For now I gotta figure out how to resize the image :-D

trolleybusses, 2021-06-01:
I'm having this issue too, except /etc/resolv.conf doesn't seem to exist at all. Creating it with a nameserver entry doesn't help either. I'm truly stumped on this.

closed 2021-06-03 by mkaczanowski