mkaczanowski/packer-plugin-builder-arm #85: Why do we need to mount `/dev/`?

daxmc99, 2021-04-20:
Came across this repo and I wanted to try out the pure docker build for MacOS. 

However, mounting `dev` with a privileged docker image is equivalent to running a script as root. 

Is this absolutely needed or is there a way to work around it?

dbast, 2021-04-20:
A free loop device (`/dev/loopX`) is required to mount images inside docker. Docker itself does not provide any loop device, so mounting `/dev` from the host makes that possible.

mkaczanowski, 2021-04-20:
@dbast explained it well. Elevating permissions is tricky and frankly speaking, the simplest option is to use a wider permission group (ie root, sudo), especially that some people use Mac, and others Linux.

If you're not comfortable with sudo/root option, you can always build stuff on top of the VM. For instance, there is a Vagrant test setup where you can do whatever you like while being isolated from the host system:
https://github.com/mkaczanowski/packer-builder-arm#troubleshooting

closed 2021-04-20 by mkaczanowski

daxmc99, 2021-04-20:
Thanks for the info! That makes sense