mkaczanowski/packer-plugin-builder-arm #103: upx-ucl is corrupting the binaries?

koenswings, 2021-05-16:
When I run the latest image published to docker hub, any build fails instantly.
Problem is the same if I build the docker image myself.
If I remove compression of the binaries (RUN upx-ucl /build/packer-builder-arm /bin/packer), it all works fine.

dbast, 2021-05-17:
Works here. The docker image got rebuild/updated 14h ago. Can you try again? thx.

koenswings, 2021-05-18:
I pulled the image again but am having the same problem. I get the running message (running /bin/packer build ...) and then it just stops without any further messages. If I build the docker image myself and remove the upx-ucl compression, things work just fine.

I am on a ubuntu-focal VirtualBox running on a Mac. I am using docker 20.10.6:

Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Build with BuildKit (Docker Inc., v0.5.1-docker)
  scan: Docker Scan (Docker Inc., v0.7.0)

Server:
 Containers: 4
  Running: 0
  Paused: 0
  Stopped: 4
 Images: 67
 Server Version: 20.10.6
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 05f951a3781f4f2c1911b05e61c160e9c30eaa8e
 runc version: 12644e614e25b05da6fd08a38ffa0cfe1903fdec
 init version: de40ad0
 Security Options:
  apparmor
  seccomp
   Profile: default
 Kernel Version: 5.4.0-73-generic
 Operating System: Ubuntu 20.04.2 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 2
 Total Memory: 981MiB
 Name: ubuntu-focal
 ID: V5LT:6F65:ANTJ:SJDK:6TT7:CJHS:SPOH:JKVN:YQKQ:DPFK:YDGD:WLJJ
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Username: koenswings
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

dbast, 2021-05-18:
Why the Virtualbox in between? On Mac with Docker installed you can directly run e.g. `docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi/raspbian.json`. Tested that and it works.

koenswings, 2021-05-18:
I know i can run it on docker-for-mac. But I had stability issues with the docker-for-mac runtime in the past and I wanted to avoid using it again. 

FYI, I do have a working setup now using my own docker build and avoiding upx-ucl compression. Just wanted to let you guys know that there are contexts where this compression might be an issue.

woa7, 2021-06-02:
i had in the past also some problems with packer compressed with upx
using lzma e.g. with 
upx-ucl --lzma
solved it for me.

mkaczanowski, 2021-06-03:
I don't have a Mac, so I can't really tell. Switching to lzma shouldn't hurt anyone and if there is a chance it fixes things for others, let's give it a shot :)

closed 2021-06-03 by mkaczanowski