mkaczanowski/packer-plugin-builder-arm #105: Error: Not PID 1 building on systemd-based OS

TheLonelyGhost, 2021-05-28:
Due to many known issues with [systemd and chroot](https://superuser.com/questions/688733/start-a-systemd-service-inside-chroot-from-a-non-systemd-based-rootfs), it seems provisioning a systemd-based operating system run into errors.

[Minimal example](https://github.com/TheLonelyGhost/raspberry-pi/tree/7aaab5a4544a1e0f03a447968694391cceaa9283).

Specifics:

```
➜ raspberry-pi % ./bin/packer build ./unifi-controller
running /bin/packer build ./unifi-controller
unifi-controller.arm.debian: output will be in this color.

==> unifi-controller.arm.debian: Retrieving rootfs_archive
==> unifi-controller.arm.debian: Retrieving rootfs_archive
==> unifi-controller.arm.debian: Trying https://raspi.debian.net/verified/20210210_raspi_4_buster.img.xz?archive=false
==> unifi-controller.arm.debian: Trying https://raspi.debian.net/verified/20210210_raspi_4_buster.img.xz?archive=false&checksum=file%3Ahttps%3A%2F%2Fraspi.debian.net%2Fverified%2F20210210_raspi_4_buster.xz.sha256
==> unifi-controller.arm.debian: https://raspi.debian.net/verified/20210210_raspi_4_buster.img.xz?archive=false&checksum=file%3Ahttps%3A%2F%2Fraspi.debian.net%2Fverified%2F20210210_raspi_4_buster.xz.sha256 => /build/packer_cache/d141fdc34e2a2516f9ae3618b30acef48a073040.xz
    unifi-controller.arm.debian: unpacking /build/packer_cache/d141fdc34e2a2516f9ae3618b30acef48a073040.xz to unifi-controller-rpi4.img
    unifi-controller.arm.debian: unpacking with custom comand: [xz --decompress /tmp/image703489848/d141fdc34e2a2516f9ae3618b30acef48a073040.xz]
    unifi-controller.arm.debian: resizing the image file unifi-controller-rpi4.img to 4G
    unifi-controller.arm.debian: expanding partition no. 2 on the image unifi-controller-rpi4.img
    unifi-controller.arm.debian: searching for empty loop device (to map unifi-controller-rpi4.img)
    unifi-controller.arm.debian: mapping image unifi-controller-rpi4.img to /dev/loop1
    unifi-controller.arm.debian: running resize2fs on /dev/loop1p2
    unifi-controller.arm.debian: mounting /dev/loop1p2 to /tmp/864532279
    unifi-controller.arm.debian: mounting /dev/loop1p1 to /tmp/864532279/boot/firmware
    unifi-controller.arm.debian: running extra setup
    unifi-controller.arm.debian: mounting /dev with: [mount --bind /dev /tmp/864532279/dev]
    unifi-controller.arm.debian: mounting /devpts with: [mount -t devpts /devpts /tmp/864532279/dev/pts]
    unifi-controller.arm.debian: mounting proc with: [mount -t proc proc /tmp/864532279/proc]
    unifi-controller.arm.debian: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/864532279/proc/sys/fs/binfmt_misc]
    unifi-controller.arm.debian: mounting sysfs with: [mount -t sysfs sysfs /tmp/864532279/sys]
    unifi-controller.arm.debian: running the provision hook
==> unifi-controller.arm.debian: Provisioning with shell script: ./scripts/hardware-img-base.sh
==> unifi-controller.arm.debian: Provisioning with shell script: ./scripts/debian-base.sh
    unifi-controller.arm.debian: Get:1 http://deb.debian.org/debian buster InRelease [121 kB]
    unifi-controller.arm.debian: Get:2 http://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
    unifi-controller.arm.debian: Get:3 http://deb.debian.org/debian buster-backports InRelease [46.7 kB]
    unifi-controller.arm.debian: Get:4 http://deb.debian.org/debian buster/main arm64 Packages [7736 kB]

# ... [snip] ...

    unifi-controller.arm.debian: Processing triggers for systemd (241-7~deb10u7) ...
    unifi-controller.arm.debian: Processing triggers for dbus (1.12.20-0+deb10u1) ...
    unifi-controller.arm.debian: Processing triggers for mime-support (3.62) ...
    unifi-controller.arm.debian: Processing triggers for ca-certificates (20200601~deb10u2) ...
    unifi-controller.arm.debian: Updating certificates in /etc/ssl/certs...
    unifi-controller.arm.debian: 0 added, 0 removed; done.
    unifi-controller.arm.debian: Running hooks in /etc/ca-certificates/update.d...
    unifi-controller.arm.debian: done.
    unifi-controller.arm.debian: Processing triggers for libgdk-pixbuf2.0-0:arm64 (2.38.1+dfsg-1) ...
==> unifi-controller.arm.debian: System has not been booted with systemd as init system (PID 1). Can't operate.
==> unifi-controller.arm.debian: Failed to create bus connection: Host is down
==> unifi-controller.arm.debian: error while running provision hook: Script exited with non-zero exit status: 1.Allowed exit codes are: [0]
    unifi-controller.arm.debian: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'unifi-controller.arm.debian' errored after 12 minutes 48 seconds: build was halted

==> Wait completed after 12 minutes 48 seconds

==> Some builds didn't complete successfully and had errors:
--> unifi-controller.arm.debian: build was halted
```

I've tried to get systemd to work inside of containers and other related headache surrounding it, but this seems to be an inherent issue with using a `chroot` provisioner. Are there other options like emulating the raspberry pi in virtualbox or specifying a provisioner like ssh in the qemu VM?

mkaczanowski, 2021-06-03:
> Are there other options like emulating the raspberry pi in virtualbox or specifying a provisioner like ssh in the qemu VM?
No, and likely won't be

closed 2021-06-03 by mkaczanowski

TheLonelyGhost, 2021-06-04:
Definitely understandable to nope out on this one. Solving it would be a nightmare of complexity for software that gets criticism for exactly this problem by numerous other projects.

I just wanted to get an issue logged for those who try to do the same thing and wonder why it's failing.

mkaczanowski, 2021-06-04:
> Solving it would be a nightmare of complexity for software

It is simply unfeasible in long run. If this plugin would support the x86 platform then I would be keener on that since the virtualization for x86 platform is quite good. ARM board emulation is currently limited, maybe you can use a limited raspberry PI image with qemu, that's all.

Regardless of virtualization state, ssh-ing to the remote box etc, is a completely different flow. Way more flaky and difficult to set up, if someone wishes to still do it I'd rather recommend using Vagrant + Vagrant provisioners