mkaczanowski/packer-plugin-builder-arm #109: arm: chroot: failed to run command '/bin/sh': Exec format error

itn3rd77, 2021-06-04:
Hi everyone,

I am stuck now for 3 days with the error message "arm: chroot: failed to run command '/bin/sh': Exec format error" and kindly asking for some help.

I am using the container from Docker Hub on my Synology NAS and trying to learn and get things running with the example raspbian-resize.json (see attachement).

[raspbian-resize.txt](https://github.com/mkaczanowski/packer-builder-arm/files/6595877/raspbian-resize.txt)

The build starts until the first shell script provisioner is executed and than fails on executing the chroot.

The error message seams to indicate that there is an architecture issue while executing the command e.g. trying to execute an arm binary from an x86 host. I am a bit lost and have no clue where to look any further.

Here is my Docker version information:

```
Client:
 Version:           18.09.8
 API version:       1.39
 Go version:        go1.11
 Git commit:        bfed4f5
 Built:             Thu Mar 18 02:58:09 2021
 OS/Arch:           linux/amd64
 Experimental:      false

Server:
 Engine:
  Version:          18.09.8
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.11
  Git commit:       c09d3c4
  Built:            Thu Mar 18 02:56:16 2021
  OS/Arch:          linux/amd64
  Experimental:     false
```

Please find the logfile with all details attached.
[packer.log](https://github.com/mkaczanowski/packer-builder-arm/files/6595867/packer.log)

Thanks for any help.

Regards
Ingo

dbast, 2021-06-04:
Does the Synology Linux Kernel support the binfmt interface (the Docker container uses that to register qemu for the binaries of various architectures)? You can check that e.g. by checking if `/proc/sys/fs/binfmt_misc` exists on the host.

See also https://www.kernel.org/doc/html/latest/admin-guide/binfmt-misc.html

itn3rd77, 2021-06-04:
Did already check this and from my point of view this is available:

```
root@server:/volume1/docker/packer/k8s-pi# ls -la /proc/sys/fs/binfmt_misc/
total 0
drwxr-xr-x 2 root root 0 Jun  4 07:41 .
dr-xr-xr-x 1 root root 0 Jun  4 07:34 ..
--w------- 1 root root 0 Jun  4 07:41 register
-rw-r--r-- 1 root root 0 Jun  4 07:41 status```
```

```
root@server:/volume1/docker/packer/k8s-pi# mount | grep binfmt
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
```

I think the part with register a new binary type (/proc/sys/fs/binfmt_misc/register) is missing. Is this something I have to do or is this done during packer-builder-arm build?

dbast, 2021-06-04:
The register stuff is done from within the Docker container, see https://github.com/mkaczanowski/packer-builder-arm/blob/master/docker/entrypoint.sh#L5 but the output is hidden.

You can also run `docker run --privileged --rm tonistiigi/binfmt --install all` and see what it prints.

itn3rd77, 2021-06-04:
Thanks for pushing me in the right direction.

The command ```docker run --privileged --rm tonistiigi/binfmt --install all ``` fails on my Synology:

```
allt@server:/volume1/docker/packer/k8s-pi# docker run --privileged --rm tonistiigi/binfmt --install
2021/06/04 11:30:10 installing: arm64 cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: ppc64le cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: s390x cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: riscv64 cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: mips64le cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: mips64 cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
2021/06/04 11:30:10 installing: arm cannot write to /proc/sys/fs/binfmt_misc/register: write /proc/sys/fs/binfmt_misc/register: invalid argument
```

With this new information I tried ```docker run --rm --privileged multiarch/qemu-user-static:register``` and that was successful.
After executing ```docker run --privileged --rm tonistiigi/binfmt``` it shows support for all emulators - great!

```
root@server:/volume1/docker/packer/k8s-pi# docker run --privileged --rm tonistiigi/binfmt
{
  "supported": [
    "linux/amd64",
    "linux/386"
  ],
  "emulators": [
    "qemu-aarch64",
    "qemu-aarch64_be",
    "qemu-alpha",
    "qemu-arm",
    "qemu-armeb",
    "qemu-hppa",
    "qemu-m68k",
    "qemu-microblaze",
    "qemu-microblazeel",
    "qemu-mips",
    "qemu-mips64",
    "qemu-mips64el",
    "qemu-mipsel",
    "qemu-mipsn32",
    "qemu-mipsn32el",
    "qemu-or1k",
    "qemu-ppc",
    "qemu-ppc64",
    "qemu-ppc64le",
    "qemu-riscv32",
    "qemu-riscv64",
    "qemu-s390x",
    "qemu-sh4",
    "qemu-sh4eb",
    "qemu-sparc",
    "qemu-sparc32plus",
    "qemu-sparc64",
    "qemu-xtensa",
    "qemu-xtensaeb"
  ]
}
```

I gave the build another try and the error ```arm: chroot: failed to run command '/bin/sh': Exec format error``` was gone, but now it simply fails with the error ```arm: chroot: failed to run command '/bin/sh': No such file or directory```

Did't expect it's getting so hard. Any further tips you can give me?

Thanks a lot!

dbast, 2021-06-04:
Does this command work `docker run --rm -t arm32v6/alpine uname -m` ... it should print `armv7l`

itn3rd77, 2021-06-04:
Ah ok. That doesn't work either ...

```
root@server:/volume1/docker/packer/k8s-pi# docker run --rm -t arm32v6/alpine uname -m
standard_init_linux.go:222: exec user process caused "no such file or directory"
```

dbast, 2021-06-04:
That means something is missing on your system... either binfmt is not properly enabled in the Synology kernel or maybe your Docker version is too old and it does not expose the binfmt stuff from the Synology host kernel to the container in the required way. So updating or using a different machine should solve this. Nothing we can fix here.

What we could do here, is to not hide the output of `/usr/bin/binfmt --install all` as part of the Docker entrypoint in order to see the errors from the beginning. That just helps to see what is the problem, but won't solve it.

itn3rd77, 2021-06-04:
Yeah ok I understand. Seams reasonable. Thanks for your time and support. Will dig a little bit deeper to understand what Synology is doing here.

closed 2021-06-04 by itn3rd77