mkaczanowski/packer-plugin-builder-arm #113: Internet problem when building

asfernandes, 2021-06-11:
I've adapted `packer/packer-builder-arm/boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json` (running with docker) to run `apt` commands but then internet does not work:

```
running /bin/packer build boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.2/release/ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz?archive=false
==> arm: Trying http://cdimage.ubuntu.com/releases/20.04.2/release/ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=file%3Ahttp%3A%2F%2Fcdimage.ubuntu.com%2Freleases%2F20.04.2%2Frelease%2FSHA256SUMS
==> arm: http://cdimage.ubuntu.com/releases/20.04.2/release/ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz?archive=false&checksum=file%3Ahttp%3A%2F%2Fcdimage.ubuntu.com%2Freleases%2F20.04.2%2Frelease%2FSHA256SUMS => /build/packer_cache/d6cf9f19f60b9a1d4a73e31d910f4d906ff2bfb5.xz
    arm: unpacking /build/packer_cache/d6cf9f19f60b9a1d4a73e31d910f4d906ff2bfb5.xz to ubuntu-20.04.img
    arm: unpacking with custom comand: [xz --decompress /tmp/image321587401/d6cf9f19f60b9a1d4a73e31d910f4d906ff2bfb5.xz]
    arm: searching for empty loop device (to map ubuntu-20.04.img)
    arm: mapping image ubuntu-20.04.img to /dev/loop13
    arm: mounting /dev/loop13p2 to /tmp/127053204
    arm: mounting /dev/loop13p1 to /tmp/127053204/boot/firmware
    arm: running extra setup
    arm: mounting /dev with: [mount --bind /dev /tmp/127053204/dev]
    arm: mounting /devpts with: [mount -t devpts /devpts /tmp/127053204/dev/pts]
    arm: mounting proc with: [mount -t proc proc /tmp/127053204/proc]
    arm: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/127053204/proc/sys/fs/binfmt_misc]
    arm: mounting sysfs with: [mount -t sysfs sysfs /tmp/127053204/sys]
    arm: running the provision hook
==> arm: Provisioning with shell script: /tmp/packer-shell266741664
==> arm: sudo: unable to resolve host ae5a3c6f3c1b: Temporary failure in name resolution
    arm: Err:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Err:2 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Err:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Reading package lists...
==> arm: W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal/InRelease  Temporary failure resolving 'ports.ubuntu.com'
==> arm: W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal-security/InRelease  Temporary failure resolving 'ports.ubuntu.com'
==> arm: W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal-updates/InRelease  Temporary failure resolving 'ports.ubuntu.com'
==> arm: W: Some index files failed to download. They have been ignored, or old ones used instead.
==> arm: sudo: unable to resolve host ae5a3c6f3c1b: Temporary failure in name resolution
    arm: Reading package lists...
    arm: Building dependency tree...
    arm: Reading state information...
    arm: The following additional packages will be installed:
    arm:   automake autotools-dev binutils binutils-aarch64-linux-gnu binutils-common
    arm:   cpp cpp-9 g++-9 gcc gcc-9 gcc-9-base icu-devtools libasan5 libatomic1
    arm:   libbinutils libc-dev-bin libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0
    arm:   libctf0 libgcc-9-dev libgomp1 libisl22 libitm1 liblsan0 libltdl-dev libmpc3
    arm:   libstdc++-9-dev libtomcrypt1 libtommath-doc libtommath1 libtsan0 libubsan1
    arm:   linux-libc-dev m4 manpages-dev
    arm: Suggested packages:
    arm:   autoconf-archive gnu-standards autoconf-doc gettext binutils-doc cpp-doc
    arm:   gcc-9-locales gcc-9-doc gcc-multilib flex bison gdb gcc-doc glibc-doc
    arm:   icu-doc libtool-doc ncurses-doc libstdc++-9-doc gfortran
    arm:   | fortran95-compiler gcj-jdk m4-doc make-doc
    arm: The following NEW packages will be installed:
    arm:   autoconf automake autotools-dev binutils binutils-aarch64-linux-gnu
    arm:   binutils-common cpp cpp-9 g++ g++-9 gcc gcc-9 gcc-9-base icu-devtools
    arm:   libasan5 libatomic1 libbinutils libc-dev-bin libc6-dev libcc1-0 libcrypt-dev
    arm:   libctf-nobfd0 libctf0 libgcc-9-dev libgomp1 libicu-dev libisl22 libitm1
    arm:   liblsan0 libltdl-dev libmpc3 libncurses-dev libstdc++-9-dev libtomcrypt-dev
    arm:   libtomcrypt1 libtommath-dev libtommath-doc libtommath1 libtool libtsan0
    arm:   libubsan1 linux-libc-dev m4 make manpages-dev zlib1g-dev
    arm: 0 upgraded, 46 newly installed, 0 to remove and 0 not upgraded.
    arm: Need to get 45.8 MB of archives.
    arm: After this operation, 201 MB of additional disk space will be used.
    arm: Err:1 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 m4 arm64 1.4.18-4
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Err:2 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 autoconf all 2.69-11.1
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Err:3 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 autotools-dev all 20180224.1
    arm:   Temporary failure resolving 'ports.ubuntu.com'
    arm: Err:4 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 automake all 1:1.16.1-4ubuntu6
[... 126 lines cut ...]
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/glibc/libc-dev-bin_2.31-0ubuntu9.2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/l/linux/linux-libc-dev_5.4.0-65.73_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libx/libxcrypt/libcrypt-dev_4.4.10-10ubuntu4_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/glibc/libc6-dev_2.31-0ubuntu9.2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/gcc-9/libstdc++-9-dev_9.3.0-17ubuntu1~20.04_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/gcc-9/g++-9_9.3.0-17ubuntu1~20.04_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/gcc-defaults/g++_9.3.0-1ubuntu2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/i/icu/icu-devtools_66.1-2ubuntu2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/i/icu/libicu-dev_66.1-2ubuntu2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libt/libtool/libltdl-dev_2.4.6-14_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/n/ncurses/libncurses-dev_6.2-0ubuntu2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libt/libtommath/libtommath1_1.2.0-3_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/libt/libtomcrypt/libtomcrypt1_1.18.2-3_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libt/libtommath/libtommath-doc_1.2.0-3_all.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libt/libtool/libtool_2.4.6-14_all.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/m/make-dfsg/make_4.2.1-1.2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/m/manpages/manpages-dev_5.05-1_all.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/z/zlib/zlib1g-dev_1.2.11.dfsg-2ubuntu1.2_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
[... cut ...]
==> arm: E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/libt/libtommath/libtommath-dev_1.2.0-3_arm64.deb  Temporary failure resolving 'ports.ubuntu.com'
==> arm: E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
==> arm: error while running provision hook: Script exited with non-zero exit status: 100.Allowed exit codes are: [0]
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'arm' errored after 2 minutes 15 seconds: build was halted
```

asfernandes, 2021-06-11:
I've added this as in `boards/raspberry-pi-4/archlinuxarm.json` and it worked.

```
    {
      "type": "shell",
      "inline": [
        "mv /etc/resolv.conf /etc/resolv.conf.bk",
        "echo 'nameserver 8.8.8.8' > /etc/resolv.conf",
      ]
    }
```

closed 2021-06-11 by asfernandes

THOM-AwS, 2021-11-21:
> I've added this as in `boards/raspberry-pi-4/archlinuxarm.json` and it worked.
> 
> ```
>     {
>       "type": "shell",
>       "inline": [
>         "mv /etc/resolv.conf /etc/resolv.conf.bk",
>         "echo 'nameserver 8.8.8.8' > /etc/resolv.conf",
>       ]
>     }
> ```

This worked for me, ta.