mkaczanowski/packer-plugin-builder-arm #115: Add example for using flasher post-processor with Docker

realies, 2021-06-17:
It seems that the flasher post-processor is not part of https://docker.io/mkaczanowski/packer-builder-arm:latest
```
error initializing post-processor 'flasher': Unknown post-processor flasher
```
Is there a way to install the plugin inside the container, or make it built-in the Docker image?

mkaczanowski, 2021-07-02:
> Is there a way to install the plugin inside the container, or make it built-in the Docker image?

you'd have to expose the plugin binary via volume. Or you can build on top of the `packer-builder-arm` image

closed 2021-07-02 by mkaczanowski

realies, 2021-07-02:
@mkaczanowski, thanks for the reply. Can we have an example, please?

mkaczanowski, 2021-07-02:
PRs are most welcome :)

realies, 2021-07-02:
@mkaczanowski to clarify, the flasher post-processor has to be built for the host architecture and the binary mounted to the docker image that runs packer-builder-arm?

realies, 2021-07-02:
@mkaczanowski, what is the path at which the plugin will get detected?