mkaczanowski/packer-plugin-builder-arm #116: Docker crashing

vincentracine, 2021-06-20:
I am trying to build a raspbian image using the pre-made docker image `mkaczanowski/packer-builder-arm` on MacOS Big Sur. The command used is below. It seems to hang for 10 seconds and then the container exists. I would expect it to build and weirdly enough, if I remove some provisioners from the `provisioners` array, then it will build fine but as soon as I try to add too many provisioners, it stops working.

`build.sh`:
```
docker run \
  --rm \
  --privileged \
  -v /dev:/dev \
  -v ${PWD}:/build \
  -e PACKER_LOG=1 \
  mkaczanowski/packer-builder-arm build -debug boards/raspbian-example.json
```

`boards/raspbian-example.json`:
```
{
  "variables": {
    "release_version": "2021.1.8",
    "wifi_name": "REMOVED",
    "wifi_password": "REMOVED",
    "hostname": "automify-sentry",
    "default_user": "REMOVED",
    "default_user_pass": "REMOVED",
    "default_user_home": "/home/pi"
  },
  "builders": [{
    "type": "arm",
    "file_urls" : ["https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip"],
    "file_checksum_url": "https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip.sha256",
    "file_checksum_type": "sha256",
    "file_target_extension": "zip",
    "image_build_method": "resize",
    "image_path": "/build/release/automifyOS-{{user `release_version`}}.img",
    "image_size": "4G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "8192",
        "filesystem": "vfat",
        "size": "256M",
        "mountpoint": "/boot"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "532480",
        "filesystem": "ext4",
        "size": "0",
        "mountpoint": "/"
      }
    ],
    "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
    "qemu_binary_source_path": "/usr/bin/qemu-arm-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-arm-static",
    "communicator": "none"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "mkdir -p /usr/local/automify/init",
        "mkdir -p /usr/local/automify/postgres-data",
        "mkdir -p /usr/local/automify/sentry-data",
        "mkdir -p /usr/local/automify/backups",
        "mkdir -p /usr/local/automify/envs",
        "mkdir -p /usr/local/automify/logs",
        "mkdir -p /usr/local/automify/updates",
        "mkdir -p {{user `default_user_home`}}/.docker",
[... 244 lines cut ...]
2021/06/20 11:53:06 packer-provisioner-file plugin: [WARN] Config file doesn't exist: /root/.packerconfig
2021/06/20 11:53:06 packer-provisioner-file plugin: [INFO] Setting cache directory: /build/packer_cache
2021/06/20 11:53:06 packer-provisioner-file plugin: args: []string{"packer-provisioner-file"}
2021/06/20 11:53:06 packer-provisioner-file plugin: Plugin address: unix /tmp/packer-plugin406249561
2021/06/20 11:53:06 Received unix RPC address for /usr/bin/packer: addr is /tmp/packer-plugin406249561
2021/06/20 11:53:06 packer-provisioner-file plugin: Waiting for connection...
2021/06/20 11:53:06 packer-provisioner-file plugin: Serving a plugin connection...
2021/06/20 11:53:06 [TRACE] Starting internal plugin packer-provisioner-file
2021/06/20 11:53:06 Starting plugin: /usr/bin/packer []string{"/usr/bin/packer", "plugin", "packer-provisioner-file"}
2021/06/20 11:53:06 Waiting for RPC address for: /usr/bin/packer
2021/06/20 11:53:08 packer-provisioner-file plugin: [INFO] Packer version: 1.7.2 [go1.16.3 linux amd64]
2021/06/20 11:53:08 packer-provisioner-file plugin: [INFO] PACKER_CONFIG env var not set; checking the default config file path
2021/06/20 11:53:08 packer-provisioner-file plugin: [INFO] PACKER_CONFIG env var set; attempting to open config file: /root/.packerconfig
2021/06/20 11:53:08 packer-provisioner-file plugin: [WARN] Config file doesn't exist: /root/.packerconfig
2021/06/20 11:53:08 packer-provisioner-file plugin: [INFO] Setting cache directory: /build/packer_cache
2021/06/20 11:53:08 packer-provisioner-file plugin: args: []string{"packer-provisioner-file"}
2021/06/20 11:53:08 packer-provisioner-file plugin: Plugin address: unix /tmp/packer-plugin534284634
2021/06/20 11:53:08 packer-provisioner-file plugin: Waiting for connection...
2021/06/20 11:53:08 Received unix RPC address for /usr/bin/packer: addr is /tmp/packer-plugin534284634
2021/06/20 11:53:08 packer-provisioner-file plugin: Serving a plugin connection...
2021/06/20 11:53:08 [TRACE] Starting internal plugin packer-provisioner-file
2021/06/20 11:53:08 Starting plugin: /usr/bin/packer []string{"/usr/bin/packer", "plugin", "packer-provisioner-file"}
2021/06/20 11:53:08 Waiting for RPC address for: /usr/bin/packer
vincent@VR-Macbook automifyOS % 
```

mkaczanowski, 2021-07-02:
Can you repro this on the vagrant / Virtualbox setup?:
https://github.com/mkaczanowski/packer-builder-arm#troubleshooting

mkaczanowski, 2021-07-06:
no response, no repro. Closing :)

closed 2021-07-06 by mkaczanowski