mkaczanowski/packer-plugin-builder-arm #120: image_size (with resize) not generated correctly

asfernandes, 2021-06-24:
I have used `"image_size": "4.5G"` (or 4.3, not sure) and the produced file has size 4617089843, which is not even multiple of 512.

mkaczanowski, 2021-07-02:
The `image_size` argument is converted to bytes via `humanize` library, so for precise size you may want to use `image_size_bytes` argument.

See attached code:
https://github.com/mkaczanowski/packer-builder-arm/blob/e19b87fcd95c7929b828f822f1693d191124f800/config/image_config.go#L52
https://github.com/mkaczanowski/packer-builder-arm/blob/e19b87fcd95c7929b828f822f1693d191124f800/builder/step_create_base_image.go#L27

closed 2021-07-02 by mkaczanowski

asfernandes, 2021-07-07:
I've now used:

```
    "image_build_method": "resize",
    "image_path": "/images/ubuntu-20.04.img",
    "image_size_bytes": "4831838208",
```

And have the error:

```
    arm: unpacking /build/packer_cache/d6cf9f19f60b9a1d4a73e31d910f4d906ff2bfb5.xz to /images/ubuntu-20.04.img
    arm: unpacking with custom comand: [xz --decompress /tmp/image398731467/d6cf9f19f60b9a1d4a73e31d910f4d906ff2bfb5.xz]
    arm: resizing the image file /images/ubuntu-20.04.img to
==> arm: error while resizing the image exit status 1: qemu-img: Parameter 'size' expects a non-negative number below 2^64
==> arm: Optional suffix k, M, G, T, P or E means kilo-, mega-, giga-, tera-, peta-
==> arm: and exabytes, respectively.
==> arm:
Build 'arm' errored after 2 minutes 10 seconds: build was halted
```

asfernandes, 2021-07-07:
The same happens with:

```
"image_size_bytes": 4831838208
```

asfernandes, 2021-07-08:
What seems to work is `image_size` with just digits:

```
"image_size": "4831838208"
```