mkaczanowski/packer-plugin-builder-arm #122: Explanation of image_partitions

khusseini, 2021-06-29:
Hello there,
I have been searching around for the difference image_partition options, however without luck.
Does anyone have a pointer to know what the different keys in the image partitions mean?
For example, what values are accepted in `type` and how to calculate the start sectors etc.
Any pointers would be apreciated

mkaczanowski, 2021-07-02:
The `image_partitions` definitions comes from `fdisk` or `parted`. Most of the ARM boards have unique requirements on where the first sector should start, what is the partition order etc. that's why we need those precise flags.

The partition setup depends on your platform. My discovery process is this:
1. download pre-made image from upstream (ie. raspbian for Raspberry PI)
2. run `fdisk -l raspbian.img`
3. use ^^ those values to replicate the partition table if needed

closed 2021-07-02 by mkaczanowski

khusseini, 2021-07-02:
Thank you @mkaczanowski 
this helps me out a lot