mkaczanowski/packer-plugin-builder-arm #130: How to build on Mac with Intel CPU?

a0s, 2021-08-19:
Hi! I am trying to build by instruction but something going wrong:

```
sudo packer build boards/odroid-xu4/ubuntu.json
arm: output will be in this color.

==> arm: Retrieving rootfs_archive
==> arm: Trying http://de.eu.odroid.in/ubuntu_18.04lts/XU3_XU4_MC1_HC1_HC2/ubuntu-18.04.1-4.14-mate-odroid-xu4-20181203.img.xz?archive=false
==> arm: Trying http://de.eu.odroid.in/ubuntu_18.04lts/XU3_XU4_MC1_HC1_HC2/ubuntu-18.04.1-4.14-mate-odroid-xu4-20181203.img.xz?archive=false&checksum=file%3Ahttp%3A%2F%2Fde.eu.odroid.in%2Fubuntu_18.04lts%2FXU3_XU4_MC1_HC1_HC2%2Fubuntu-18.04.1-4.14-mate-odroid-xu4-20181203.img.xz.md5sum
==> arm: http://de.eu.odroid.in/ubuntu_18.04lts/XU3_XU4_MC1_HC1_HC2/ubuntu-18.04.1-4.14-mate-odroid-xu4-20181203.img.xz?archive=false&checksum=file%3Ahttp%3A%2F%2Fde.eu.odroid.in%2Fubuntu_18.04lts%2FXU3_XU4_MC1_HC1_HC2%2Fubuntu-18.04.1-4.14-mate-odroid-xu4-20181203.img.xz.md5sum => /Users/a0s/3dprinting/packer-builder-arm/packer_cache/21d459da00cabdca71bd375bb18a3a46f7c67b96.xz
    arm: unpacking /Users/a0s/3dprinting/packer-builder-arm/packer_cache/21d459da00cabdca71bd375bb18a3a46f7c67b96.xz to odroid-xu4.img
    arm: unpacking with custom comand: [xz -d /tmp/image210140239/21d459da00cabdca71bd375bb18a3a46f7c67b96.xz]
    arm: searching for empty loop device (to map odroid-xu4.img)
==> arm: error losetup -f exec: "losetup": executable file not found in $PATH:
==> arm: error while unmounting loop device exec: "losetup": executable file not found in $PATH:
Build 'arm' errored after 1 minute 5 seconds: build was halted

==> Wait completed after 1 minute 5 seconds

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
```

Am i missed something? `brew install losetup` get nothing. What additional dependencies i need to install?

Also, i tried build with docker (i have Docker on Mac installed):

```
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm build boards/odroid-xu4/ubuntu.json

uname -a:  Linux c53510389cfa 5.10.25-linuxkit #1 SMP Tue Mar 23 09:27:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
installing: riscv64 qemu-riscv64 already registered
installing: mips64le qemu-mips64el already registered
installing: mips64 qemu-mips64 already registered
{
  "supported": [
    "linux/amd64",
    "linux/arm64",
    "linux/ppc64le",
    "linux/s390x",
    "linux/386",
    "linux/arm/v7",
    "linux/arm/v6"
  ],
  "emulators": [
    "qemu-aarch64",
    "qemu-arm",
    "qemu-mips64",
    "qemu-mips64el",
    "qemu-ppc64le",
    "qemu-riscv64",
    "qemu-s390x"
  ]
}
running /bin/packer build boards/odroid-xu4/ubuntu.json
Error: Failed to initialize build "arm"

error initializing builder 'arm': fork/exec /build/packer-builder-arm: exec
format error

==> Wait completed after 35 microseconds

==> Builds finished but no artifacts were created.
```

Any help would be appreciate!

mkaczanowski, 2021-12-27:
I don't have a Mac, so can't help here. Depending on the complexity I'd go with a special handlers/steps for MacOS X (it could be as simple as finding equivalent tooling on mac or as hard as changing the whole flow).

mikemorris, 2021-12-28:
> A fourth option might be to use a privileged docker container for this.

I understand why this won't run natively on macOS due to the dependency on `losetup`, but I'm a little unclear on why the [privileged Docker container build documented in the README](https://github.com/mkaczanowski/packer-builder-arm#usage-via-local-container-build-supports-amd64aarch64-hosts) fails with

> Error: Failed to initialize build "arm"

> error initializing builder 'arm': fork/exec /build/packer-builder-arm: exec
format error

Might that be a simpler issue to fix? I'm actually hitting this exact same error with the Docker and Vagrant builds on https://github.com/solo-io/packer-plugin-arm-image too. Is it perhaps an issue with a newer version of Packer config syntax, or a need to upgrade these plugins to use the new plugin SDK for Packer v1.7 and later?

mikemorris, 2022-02-12:
Welp, think I figured this out. When using the Docker build, it appears the local directory is not linked into the Docker container (which is probably fine/expected), but this means the plugin binary _must_ be installed to the `.packer.d/plugins` directory following the steps in `manually (multi-component plugin) at https://www.packer.io/plugins#installing-plugins

```
go build
mv packer-builder-arm ~/.packer.d/plugins
```

kevinelliott, 2022-02-17:
> Welp, think I figured this out. When using the Docker build, it appears the local directory is not linked into the Docker container (which is probably fine/expected), but this means the plugin binary _must_ be installed to the `.packer.d/plugins` directory following the steps in `manually (multi-component plugin)` at https://www.packer.io/plugins#installing-plugins for Packer to find a plugin providing `arm` build support.
> 
> ```
> go build
> mv packer-builder-arm ~/.packer.d/plugins
> ```
> 
> Adding support for the Packer 1.7 `init` command as described in #100 might help avoid this confusion.
> 
> Actually, looking more closely at the Dockerfile, this might be an issue with a newer version of Go or specific to macOS -
> 
> https://github.com/mkaczanowski/packer-builder-arm/blob/0eff8b7c5475cdca23dd28c2eba174df97585bf7/docker/Dockerfile#L52
> 
> assumes the binary will be built into `./build` - running `go build` on my machine instead puts the binary into the repo root.

Can you elaborate on what to do for Mac using Docker? I'm running into this issue and struggling to figure out how to get things to work.

cristian-rivera, 2022-08-06:
> Can you elaborate on what to do for Mac using Docker? I'm running into this issue and struggling to figure out how to get things to work.

This is working for me on an M1 Mac

```
git clone https://github.com/mkaczanowski/packer-builder-arm
cd packer-builder-arm
go mod download
go build
mv packer-builder-arm ./build/packer-builder-arm
docker build -t packer-builder-arm -f docker/Dockerfile .
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi/raspbian.json
```

`mv packer-builder-arm ./build/packer-builder-arm` is the extra necessary command to make this work. Simply running `go build` will place the binary at `./packer-builder-arm`, but the Docker image expects it to be in `./build/packer-builder-arm`.

goshlanguage, 2022-09-19:
> This is working for me on an M1 Mac
>
> ```
> git clone https://github.com/mkaczanowski/packer-builder-arm
> cd packer-builder-arm
> go mod download
> go build
> mv packer-builder-arm ./build/packer-builder-arm [...]

@cristian-rivera I am curious what your container runtime is?

I am trying to replicate but can not. On an x86 mac, running colima, oddly either with x86 virt or arm64 virt:

`x86`:

```
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi/raspbian.json

uname -a:  Linux a29b632e5c6e 5.10.93-0-virt #1-Alpine SMP Thu, 27 Jan 2022 09:34:38 +0000 x86_64 x86_64 x86_64 GNU/Linux
installing: mips64 OK
installing: mips64le OK
{
  "supported": [
    "linux/amd64",
    "linux/arm64",
    "linux/riscv64",
    "linux/ppc64le",
    "linux/s390x",
    "linux/386",
    "linux/mips64le",
    "linux/mips64",
    "linux/arm/v7",
    "linux/arm/v6"
  ],
  "emulators": [
    "qemu-aarch64",
    "qemu-arm",
    "qemu-i386",
    "qemu-i486",
    "qemu-mips64",
    "qemu-mips64el",
    "qemu-ppc64le",
    "qemu-riscv64",
    "qemu-s390x"
  ]
}
running /bin/packer build boards/raspberry-pi/raspbian.json
Error: Failed to initialize build "arm"

error initializing builder 'arm': fork/exec /build/packer-builder-arm: exec
format error

==> Wait completed after 59 microseconds

==> Builds finished but no artifacts were created.
```

`arm64`:

```sh
docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build packer-builder-arm build boards/raspberry-pi/raspbian.json
uname -a:  Linux c8e005bcaf75 5.10.109-0-virt #1-Alpine SMP Mon, 28 Mar 2022 11:20:52 +0000 aarch64 aarch64 aarch64 GNU/Linux
{
  "supported": [
    "linux/arm64",
    "linux/amd64",
    "linux/riscv64",
    "linux/ppc64le",
    "linux/s390x",
    "linux/386",
    "linux/mips64le",
    "linux/mips64",
    "linux/arm/v7",
    "linux/arm/v6"
  ],
  "emulators": [
    "qemu-i386",
    "qemu-mips64",
    "qemu-mips64el",
    "qemu-ppc64le",
    "qemu-riscv64",
    "qemu-s390x",
    "qemu-x86_64"
  ]
}
running /bin/packer build boards/raspberry-pi/raspbian.json
Error: Failed to initialize build "arm"

error initializing builder 'arm': fork/exec /build/packer-builder-arm: exec
format error

==> Wait completed after 273 microseconds

==> Builds finished but no artifacts were created.
```

Do you perhaps get an `arm/v8` in your `supported` list on your runs?

dbast, 2022-10-19:
There have been bugs regarding the Docker container not really be multi-arch in the past (wrong base go image), but those have been fixed some time ago. Tried the example of the README and this works here on multiple M1 MACs:
`docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm:1.0.2 build boards/raspberry-pi/raspbian.json` 

Can you all try that and report back with log in case there are still issues?

dbast, 2022-10-23:
Spent more time with above mentioned ideas and proposed solutions and compiled a README usage update https://github.com/mkaczanowski/packer-builder-arm/pull/232 to explain and avoid/prevent the error `error initializing builder 'arm': fork/exec /build/packer-builder-arm: exec format error`.

closed 2022-10-23 by dbast