mkaczanowski/packer-plugin-builder-arm #144: Mount /etc/resolv.conf during extra setup

Lykos153, 2021-10-17:
Before chrooting, `packer-builder-arm` mounts some additional directories which are necessary for the chroot environment to work. `/etc/resolv.conf`, however, is missing, resulting in network problems down the line, as can be seen in various tickets: #16 #68 #83 #113

Please, during extra setup, mount the host's `resolv.con`

mkaczanowski, 2021-11-16:
I would expect it should work out of the box, but having someone to test it out would be great

mkaczanowski, 2021-12-27:
For my own reference:
* https://www.ctrl.blog/entry/resolvconf-tutorial.html
* `sudo systemd-nspawn --bind-ro=/etc/resolv.conf -D /tmp/4232331700 -M machine`

chrsoo, 2022-01-04:
@mkaczanowski wrote:
> I guess it should be as simple as:
> 
>     mount --bind /etc/resolv.conf /.../etc/resolv.conf
Well... if that was the case could we not just use the `additional_chroot_mounts` in the packer file? I tried and it results in the following error:

    ==> arm: error creating mount directory: mkdir /tmp/561410291/etc/resolv.conf: file exists

... which I don't think is that surprising given that we are talking about a file or link and not a directory.

Also the [BasicChroot](https://help.ubuntu.com/community/BasicChroot#Setting-up_the_chroot) Ubuntu article suggest copying the file.

I tried to implement a PR for this but my Go skills and knowledge of the codebase is not up to par it would seem as I can't make it work... I always get errors along the lines of

    ==> arm: error while creating destination: open /tmp/910554841/etc/resolv.conf: no such file or directory: '/tmp/910554841/etc/resolv.conf'

... which I don't quite understand as the upstream base images OS filesystem *should* be available in the temporary directory after setting up the chroot mounts... 

I have tried patching the code base with the following snippet in several of the steps but always with the same result; perhaps you can spot the error?

```go
src := "/etc/resolv.conf"
dst := filepath.Join(imageMountpoint, src)
source, err := os.Open(src)
if err != nil {
	ui.Error(fmt.Sprintf("error while opening source: %v: '%s'", err, src))
	return multistep.ActionHalt
}
defer source.Close()

// Should we backup the /etc/resolv.conf if it exist and restore it before creating the final image?
destination, err := os.Create(dst)
if err != nil {
	ui.Error(fmt.Sprintf("error while creating destination: %v: '%s'", err, dst))
	return multistep.ActionHalt
}
defer destination.Close()

_, err = io.Copy(destination, source)
if err == nil {
	ui.Message(fmt.Sprintf("copied file from '%s' to '%s'", src, dst))
} else {
	ui.Error(fmt.Sprintf("error while copying: %v: from '%s' to '%s'", err, src, dst))
	return multistep.ActionHalt
}
```
**If you are able to provide a branch with a fix I will happily test it on my M1 Powerbook as the issue currently is blocking me from using the the provisioner with ubuntu-20.04 images!**

chrsoo, 2022-01-04:
Maybe the problem is that `/tmp/.../etc/resolv.conf` on the mounted image is a symlink that points to nowhere? Then I guess deleting it before copying should resolve the issue...

michalfita, 2023-06-02:
That symlink doesn't point to nowhere, but to a file that systemd manages on boot - my understanding is it's connected with Network Manager, so DNS addresses obtained from DHCP during roaming between networks work smoothly.

I'm doing this in my provisionings:
```
      "mkdir -p /run/systemd/resolve",
      "echo 'nameserver 1.1.1.1' > /run/systemd/resolve/stub-resolv.conf",
```

closed 2026-09-23 by dbast