mkaczanowski/packer-plugin-builder-arm #168: Name resolution failure within chroot

LocalAreaNitwit, 2022-01-31:
Hi there,

I expect this is something to do with my environment but thought I'd reach out to see if there's a solution.

Regardless of what nameservers I put in resolv.conf DNS lookup required for my build do no function. Using the Raspberry-Pi 4 build as a base I'm unable to get passed the `pacman -Syu --noconfirm --needed` step within the shell provisioner.

I've put a cat after the "echo 'nameserver 1.1.1.1' > /etc/resolv.conf" line to confirm the file is being populated correctly and it is. 

```
==> arm.arch_linux_arm: Provisioning with shell script: /tmp/packer-shell2360027272
    arm.arch_linux_arm: nameserver 1.1.1.1
==> arm.arch_linux_arm: gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
==> arm.arch_linux_arm: gpg: no ultimately trusted keys found
==> arm.arch_linux_arm: gpg: starting migration from earlier GnuPG versions
==> arm.arch_linux_arm: gpg: porting secret keys from '/etc/pacman.d/gnupg/secring.gpg' to gpg-agent
==> arm.arch_linux_arm: gpg: migration succeeded
    arm.arch_linux_arm: ==> Generating pacman master key. This may take some time.
==> arm.arch_linux_arm: gpg: Generating pacman keyring master key...
==> arm.arch_linux_arm: gpg: key ED0459306C96AB66 marked as ultimately trusted
==> arm.arch_linux_arm: gpg: directory '/etc/pacman.d/gnupg/openpgp-revocs.d' created
==> arm.arch_linux_arm: gpg: revocation certificate stored as '/etc/pacman.d/gnupg/openpgp-revocs.d/22AD9318ED98BC4ACA3CCD4BED0459306C96AB66.rev'
==> arm.arch_linux_arm: gpg: Done
    arm.arch_linux_arm: ==> Updating trust database...
==> arm.arch_linux_arm: gpg: marginals needed: 3  completes needed: 1  trust model: pgp
==> arm.arch_linux_arm: gpg: depth: 0  valid:   1  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 1u
    arm.arch_linux_arm: ==> Appending keys from archlinuxarm.gpg...
    arm.arch_linux_arm: ==> Locally signing trusted keys in keyring...
    arm.arch_linux_arm:   -> Locally signed 3 keys.
    arm.arch_linux_arm: ==> Importing owner trust values...
==> arm.arch_linux_arm: gpg: setting ownertrust to 4
==> arm.arch_linux_arm: gpg: inserting ownertrust of 4
==> arm.arch_linux_arm: gpg: setting ownertrust to 4
    arm.arch_linux_arm: ==> Updating trust database...
==> arm.arch_linux_arm: gpg: marginals needed: 3  completes needed: 1  trust model: pgp
==> arm.arch_linux_arm: gpg: depth: 0  valid:   1  signed:   3  trust: 0-, 0q, 0n, 0m, 0f, 1u
==> arm.arch_linux_arm: gpg: depth: 1  valid:   3  signed:   1  trust: 0-, 0q, 0n, 3m, 0f, 0u
==> arm.arch_linux_arm: gpg: depth: 2  valid:   1  signed:   0  trust: 1-, 0q, 0n, 0m, 0f, 0u
    arm.arch_linux_arm: :: Synchronizing package databases...
    arm.arch_linux_arm:  core downloading...
==> arm.arch_linux_arm: error: failed retrieving file 'core.db' from mirror.archlinuxarm.org : Could not resolve host: mirror.archlinuxarm.org
==> arm.arch_linux_arm: warning: too many errors from mirror.archlinuxarm.org, skipping for the remainder of this transaction
==> arm.arch_linux_arm: error: failed to synchronize all databases (invalid url for server)
==> arm.arch_linux_arm: error while running provision hook: Script exited with non-zero exit status: 1.Allowed exit codes are: [0]
==> arm.arch_linux_arm: optional `fuser -k` failed with exit status 1:
Build 'arm.arch_linux_arm' errored after 22 seconds 727 milliseconds: build was halted

==> Wait completed after 22 seconds 727 milliseconds

==> Some builds didn't complete successfully and had errors:
--> arm.arch_linux_arm: build was halted

==> Builds finished but no artifacts were created.
```

Any ideas? 
Thanks in advance!

cdecoux, 2022-02-18:
https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/archlinuxarm.json#L40

Did you try it exactly like this? Including using the Google DNS nameserver?

LocalAreaNitwit, 2022-03-05:
I did indeed. I've just tried again with zero changes since my last try (other than system updates etc.) and it appears to be working. Must have been something to do with my hosts networking. Thanks!

closed 2022-03-05 by LocalAreaNitwit