mkaczanowski/packer-plugin-builder-arm #199: builder from ISO file

LIUQyou, 2022-06-12:
Hi,

I am working on a gem5 project, and I need to create an image file from the arm server iso.

I am not sure is there any method I can build an image from an iso file? 

thanks in advance.

best regards,
Qunyou Liu

LIUQyou, 2022-06-16:
The following is the JSON I used to test:
```
{
  "variables": {},
  "builders": [{
    "type": "arm",
    "file_urls" : ["http://cdimage.ubuntu.com/releases/22.04/release/ubuntu-22.04-live-server-arm64.iso"],
    "file_checksum_url": "http://cdimage.ubuntu.com/releases/22.04/release/SHA256SUMS",
    "file_checksum_type": "sha256",
    "file_target_extension": "iso",
    "image_build_method": "reuse",
    "image_path": "ubuntu-22.04.img",
    "image_size": "5.0G",
    "image_type": "dos",
    "image_partitions": [
      {
        "name": "boot",
        "type": "c",
        "start_sector": "2048",
        "filesystem": "fat",
        "size": "256M",
        "mountpoint": "/boot/firmware"
      },
      {
        "name": "root",
        "type": "83",
        "start_sector": "526336",
        "filesystem": "ext4",
        "size": "2.8G",
        "mountpoint": "/"
      }

    ],
    "image_chroot_env": ["PATH=/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"],
    "qemu_binary_source_path": "/usr/bin/qemu-aarch64-static",
    "qemu_binary_destination_path": "/usr/bin/qemu-aarch64-static"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "touch /tmp/test"
      ]
    }
  ],
  "post-processors": []
}
```
But I encounter problem like the following:
```
==> arm: http://cdimage.ubuntu.com/releases/22.04/release/ubuntu-22.04-live-server-arm64.iso?archive=false&checksum=file%3Ahttp%3A%2F%2Fcdimage.ubuntu.com%2Freleases%2F22.04%2Frelease%2FSHA256SUMS => /home/quliu/.cache/packer/d4ed43e4fa8753e19c2bbf7e483e7ce1663d5312.iso
    arm: using raw image
    arm: mapping image ubuntu-22.04.img to free loopback device
    arm: image ubuntu-22.04.img mapped to /dev/loop26
    arm: mounting /dev/loop26p2 to /tmp/2090395387
    arm: mounting /dev/loop26p1 to /tmp/2090395387/boot/firmware
==> arm: exit status 32
==> arm: exit status 32
2022/06/16 10:43:56 [INFO] (telemetry) ending arm
==> Wait completed after 3 seconds 859 milliseconds
2022/06/16 10:43:56 machine readable: error-count []string{"1"}
==> Some builds didn't complete successfully and had errors:
2022/06/16 10:43:56 machine readable: arm,error []string{"build was halted"}
Build 'arm' errored after 3 seconds 859 milliseconds: build was halted

==> Builds finished but no artifacts were created.
2022/06/16 10:43:56 [INFO] (telemetry) Finalizing.
==> Wait completed after 3 seconds 859 milliseconds

==> Some builds didn't complete successfully and had errors:
--> arm: build was halted

==> Builds finished but no artifacts were created.
2022/06/16 10:43:57 waiting for all plugin processes to complete...
2022/06/16 10:43:57 /home/quliu/Desktop/packer-builder-arm/packer-builder-arm: plugin process exited
2022/06/16 10:43:57 /usr/bin/packer: plugin process exited
```

Thanks

dbast, 2022-10-30:
The mentioned iso dos not contain a writeable filesystem where it is possible to chroot into. You can instead use `http://cdimage.ubuntu.com/releases/20.04.2/release/ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz`, where this is possible, see also https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json

closed 2022-10-30 by dbast