mkaczanowski/packer-plugin-builder-arm #215: Building RPI Image using Gitlab CI - Docker Image

marcelomanzo, 2022-09-08:
Hi,

I am struggling to generate an rpi image using Gitlab CI - Docker Image, please note that works fine using wls - ubuntu.

1) Debian Docker Image:
build_os_dist:
  image: "debian:bookworm-slim"
  stage: .pre
  script:
    - apt-get update
    - apt-get -y install golang unzip qemu-user-static e2fsprogs dosfstools wget git qemu-utils parted e2fsprogs
    - wget https://releases.hashicorp.com/packer/1.8.3/packer_1.8.3_linux_amd64.zip
    - unzip packer_1.8.3_linux_amd64.zip
    - mv packer /usr/local/bin/
    - rm packer_1.8.3_linux_amd64.zip
    - git clone https://github.com/mkaczanowski/packer-builder-arm
    - cd packer-builder-arm
    - go mod download
    - go build
    - packer build magicbox-os_arm64.json
  tags:
    - linux
  artifacts:
    expire_in: 1 hr
    paths:
      - os_dist/*.zip

Result:
[image]

I am wondering if you have any suggestion.

marcelomanzo, 2022-09-08:
Same issue when using docker:
build_os_dist:
  image: "docker:latest"
  stage: .pre
  services:
    - docker:dind
  script:
    - cd os_dist
    - docker run --rm --privileged -v /dev:/dev -v ${PWD}:/build mkaczanowski/packer-builder-arm build magicbox-os_arm64.json
  tags:
    - linux
  artifacts:
    expire_in: 1 hr
    paths:
      - os_dist/*.zip

[image]

marcelomanzo, 2022-09-08:
if I use reuse instead of resize, the error message changes:
arm: unpacking /root/.cache/packer/1cd614d18502c032fe64e3[91](https://gitlab.com/simracingstudio/srsapp/-/jobs/3000622274#L91)454d304d9d2b5a0a.xz to magicbox-os-arm64.img
    arm: unpacking with custom command: [xz -d /tmp/image555151982/1cd614d18502c032fe64e391454d304d9d2b5a0a.xz]
    arm: mapping image magicbox-os-arm64.img to free loopback device
    arm: image magicbox-os-arm64.img mapped to /dev/loop0
    arm: mounting /dev/loop0p2 to /tmp/7078596[93](https://gitlab.com/simracingstudio/srsapp/-/jobs/3000622274#L93)
==> arm: exit status 32
==> arm: exit status 32
==> arm: exit status 32
Build 'arm' errored after 1 minute 7 seconds: build was halted
==> Wait completed after 1 minute 7 seconds
==> Some builds didn't complete successfully and had errors:
--> arm: build was halted
==> Builds finished but no artifacts were created.
Cleaning up project directory and file based variables
00:01
ERROR: Job failed: exit code 1

ThijsVredenbregt, 2024-08-02:
Did you figure this out ?

dbast, 2026-09-23:
Fixed by  #368

closed 2026-09-23 by dbast