mkaczanowski/packer-plugin-builder-arm #245: File not found error for script provided to shell provisioner

benalexau, 2022-11-30:
I use GitHub Actions to build an `ArchLinuxARM-rpi-armv7-latest.tar.gz` image using a file-based shell provisioner configured as follows:

```
    {
      "type": "shell",
      "scripts": [
        "build.sh"
      ]
    }
```

A file not found error is reported after the shell script is uploaded:

```
==> arm: Uploading resizerootfs => /tmp/resizerootfs
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Uploading to chroot dir: /tmp/3848570873/tmp/resizerootfs
2022/11/29 23:27:20 packer-provisioner-file plugin: [INFO] 449 bytes written for 'uploadData'
2022/11/29 23:27:20 [INFO] 449 bytes written for 'uploadData'
2022/11/29 23:27:20 packer-provisioner-file plugin: closing
2022/11/29 23:27:20 closing
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 closing
2022/11/29 23:27:20 [INFO] (telemetry) ending file
2022/11/29 23:27:20 [INFO] (telemetry) Starting provisioner shell
==> arm: Provisioning with shell script: build.sh
2022/11/29 23:27:20 packer-provisioner-shell plugin: Opening build.sh for reading
2022/11/29 23:27:20 packer-provisioner-shell plugin: [INFO] 2736 bytes written for 'uploadData'
2022/11/29 23:27:20 [INFO] 2736 bytes written for 'uploadData'
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Uploading to chroot dir: /tmp/3848570873/tmp/script_5870.sh
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Executing: /bin/sh []string{"/bin/sh", "-c", "chroot /tmp/3848570873 /bin/sh -c \"chmod 0755 /tmp/script_5870.sh\""}
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Chroot execution exited with '127': '"chmod 0755 /tmp/script_5870.sh"'
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 [INFO] RPC endpoint: Communicator ended with: 127
2022/11/29 23:27:20 [INFO] RPC client: Communicator ended with: 127
2022/11/29 23:27:20 [INFO] RPC endpoint: Communicator ended with: 127
2022/11/29 23:27:20 packer-provisioner-shell plugin: [INFO] RPC client: Communicator ended with: 127
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Executing: /bin/sh []string{"/bin/sh", "-c", "chroot /tmp/3848570873 /bin/sh -c \"chmod +x /tmp/script_5870.sh; PACKER_BUILDER_TYPE='arm' PACKER_BUILD_NAME='arm'  /tmp/script_5870.sh\""}
==> arm: -c: chmod +x /tmp/script_5870.sh; PACKER_BUILDER_TYPE='arm' PACKER_BUILD_NAME='arm'  /tmp/script_5870.sh: No such file or directory
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 Chroot execution exited with '127': '"chmod +x /tmp/script_5870.sh; PACKER_BUILDER_TYPE='arm' PACKER_BUILD_NAME='arm'  /tmp/script_5870.sh"'
2022/11/29 23:27:20 packer-builder-arm plugin: 2022/11/29 23:27:20 [INFO] RPC endpoint: Communicator ended with: 127
2022/11/29 23:27:20 [INFO] 0 bytes written for 'stdout'
2022/11/29 23:27:20 [INFO] 132 bytes written for 'stderr'
2022/11/29 23:27:20 [INFO] RPC client: Communicator ended with: 127
2022/11/29 23:27:20 [INFO] RPC endpoint: Communicator ended with: 127
2022/11/29 23:27:20 packer-provisioner-shell plugin: [INFO] 0 bytes written for 'stdout'
2022/11/29 23:27:20 packer-provisioner-shell plugin: [INFO] 132 bytes written for 'stderr'
2022/11/29 23:27:20 packer-provisioner-shell plugin: [INFO] RPC client: Communicator ended with: 127
2022/11/29 23:27:20 [INFO] (telemetry) ending shell
==> arm: error while running provision hook: Script exited with non-zero exit status: 127. Allowed exit codes are: [0]
    arm: optional (please ignore) `fuser -k` failed with exit status 1:
    arm: unmounting /tmp/3848570873/sys
    arm: unmounting /tmp/3848570873/proc/sys/fs/binfmt_misc
    arm: unmounting /tmp/3848570873/proc
    arm: unmounting /tmp/3848570873/dev/pts
    arm: unmounting /tmp/3848570873/dev
    arm: unmounting /tmp/3848570873/boot-tmp
    arm: unmounting /tmp/3848570873
2022/11/29 23:27:23 [INFO] (telemetry) ending arm
Build 'arm' errored after 24 seconds 814 milliseconds: build was halted
==> Wait completed after 24 seconds 814 milliseconds
```

Please see here for the full [JSON configuration](https://github.com/benalexau/archlinux-pi/blob/main/archlinux-pi-rp4.json) and GitHub Actions [workflow](https://github.com/benalexau/archlinux-pi/blob/main/.github/workflows/pi4.yaml).

Possibly related to #244?

dbast, 2022-11-30:
@benalexau This fixes it https://github.com/benalexau/archlinux-pi/pull/1 see this run https://github.com/dbast/archlinux-pi/actions/runs/3582220030/jobs/6026188780

Not sure yet what the cause is. Currently comparing various build logs and code snippets:
* https://github.com/tonistiigi/binfmt/blob/master/Dockerfile#L126-L134
* https://github.com/mkaczanowski/packer-builder-arm/blob/master/docker/Dockerfile#L52
* https://github.com/mkaczanowski/packer-builder-arm/blob/master/docker/entrypoint.sh#L7

Ideas welcome. Closing as duplicate of https://github.com/mkaczanowski/packer-builder-arm/issues/244

closed 2022-11-30 by dbast

benalexau, 2022-11-30:
Thanks @dbast, much appreciated.