mkaczanowski/packer-plugin-builder-arm #251: Is there an example how to create

michilehr, 2022-12-15:
Hi,

is there an example how to identify the values for `image_partitions`, `image_type` and `image_size` when creating a new config for another board and distribution?

For example, I want to create an image for the [Rock 4 SE](https://wiki.radxa.com/Rock4/downloads) with [Debian11 Desktop(Linux 4.4)](https://github.com/radxa/debos-radxa/releases/download/20221109-1007/rockpi-4b-debian-bullseye-xfce4-arm64-20221109-1447-gpt.img.xz).

Thank you for your help!

dbast, 2022-12-16:
Hi @michilehr,

packer-build-arm can operate in two modes:
* creating a new target image based on serval parameters (image_partitions, image_type and image_size) AND an archive of a rootfs (often a tarball of the root file system). That can be seen e.g. [here](https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/archlinuxarm.pkr.hcl#L11) ... this seems to be not applicable in the above mentioned Rock 4 case as the link points to a disk image (not a rootfs archive).
* reusing an existing image: applicable here, as referenced link above seems to be an entire disk image, containing a partition table and potentially multiple partitions. An *reuse* example can be seen [here](https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi-4/ubuntu_server_20.04_arm64.pkr.hcl#L22). Even though no partitions are created, you still have to define them, so that they get correctly mounted during the packer run. It should be possible to find the correct values via e.g.:

  ```
  wget https://github.com/radxa/debos-radxa/releases/download/20221109-1007/rockpi-4b-debian-bullseye-xfce4-arm64-20221109-1447-gpt.img.xz
  xz -d rockpi-4b-debian-bullseye-xfce4-arm64-20221109-1447-gpt.img.xz
  fdisk -l rockpi-4b-debian-bullseye-xfce4-arm64-20221109-1447-gpt.img
  ```

  To find out the correct `mountpoint`s you maybe have to look into `/etc/fstab` inside the image (can be also done via guessing and a packer run).

dbast, 2022-12-28:
```
/sbin/fdisk -l rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img 
Disk rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img: 3.73 GiB, 4000000000 bytes, 7812500 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 61B4DE28-19A0-4F27-9B51-CFA5E74F50B5

Device                                                         Start     End Sectors  Size Type
rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img1      64    8063    8000  3.9M Microsoft basic data
rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img2   16384   24575    8192    4M Microsoft basic data
rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img3   24576   32767    8192    4M Microsoft basic data
rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img4   32768 1081343 1048576  512M EFI System
rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img5 1081344 7812466 6731123  3.2G Linux filesystem
```

closed 2022-12-28 by dbast

michilehr, 2022-12-29:
> fdisk
> 
> ```
> /sbin/fdisk -l rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img 
> Disk rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img: 3.73 GiB, 4000000000 bytes, 7812500 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: gpt
> Disk identifier: 61B4DE28-19A0-4F27-9B51-CFA5E74F50B5
> 
> Device                                                         Start     End Sectors  Size Type
> rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img1      64    8063    8000  3.9M Microsoft basic data
> rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img2   16384   24575    8192    4M Microsoft basic data
> rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img3   24576   32767    8192    4M Microsoft basic data
> rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img4   32768 1081343 1048576  512M EFI System
> rockpi-4b-debian-bullseye-xfce4-arm64-20221201-1203-gpt.img5 1081344 7812466 6731123  3.2G Linux filesystem
> ```
> 
> blkid
> 
> ```
> blkid
> /dev/mapper/loop0p1: PARTLABEL="loader1" PARTUUID="efb172eb-6ac9-4a1b-8757-04fd3927a53f"
> /dev/mapper/loop0p2: PARTLABEL="loader2" PARTUUID="0c97b1a7-175b-4aa9-b8bd-3e6d8b5c2bfa"
> /dev/mapper/loop0p3: PARTLABEL="trust" PARTUUID="dac28776-fec9-479a-871b-ebdd1e52c608"
> /dev/mapper/loop0p4: LABEL_FATBOOT="boot" LABEL="boot" UUID="BBE6-9685" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="boot" PARTUUID="3537e5c1-6e9e-4844-9d77-ce4bbf3e974f"
> /dev/mapper/loop0p5: LABEL="rootfs" UUID="40a30c9d-171e-4acc-9531-330565ce4f88" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="rootfs" PARTUUID="f59a4dc9-2c89-49c7-a767-a2a7f7756146"
> ```
> 
> fstab
> 
> ```
> /etc/fstab 
> UUID=40a30c9d-171e-4acc-9531-330565ce4f88	/	ext4	defaults	0	1
> UUID=BBE6-9685	/boot	vfat	defaults,x-systemd.automount	0	2
> ```

Thanks a lot @dbast 🙏