mkaczanowski/packer-plugin-builder-arm #258: Failed to find binfmt_misc for qemu-arm under /proc/sys/fs/binfmt_misc

shanduur, 2023-01-15:
I am facing this issue as in the title.

The environment:

## Packer:

```
$ packer version
Packer v1.8.5
```

## Kernel

```
$ uname -a
Linux rocky.dev 4.18.0-372.19.1.el8_6.x86_64 #1 SMP Tue Aug 2 16:19:42 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

## OS

```
$ cat /etc/os-release
NAME="Rocky Linux"
VERSION="8.5 (Green Obsidian)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="8.5"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Rocky Linux 8.5 (Green Obsidian)"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:rocky:rocky:8:GA"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
ROCKY_SUPPORT_PRODUCT="Rocky Linux"
ROCKY_SUPPORT_PRODUCT_VERSION="8"
```

## Installed Packages

```
$ sudo dnf list installed | grep qemu
ipxe-roms-qemu.noarch                             20200823-7.git4bd064de.el8                                @raven-extras          
libvirt-daemon-driver-qemu.x86_64                 8.2.0-1.el8                                               @raven-extras          
qemu-audio-alsa.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-audio-jack.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-audio-oss.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-audio-pa.x86_64                              15:7.0.0-2.el8                                            @raven-extras          
qemu-audio-sdl.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-audio-spice.x86_64                           15:7.0.0-2.el8                                            @raven-extras          
qemu-block-curl.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-block-dmg.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-block-gluster.x86_64                         15:7.0.0-2.el8                                            @raven-extras          
qemu-block-iscsi.x86_64                           15:7.0.0-2.el8                                            @raven-extras          
qemu-block-rbd.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-block-ssh.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-char-baum.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-char-spice.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-common.x86_64                                15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-qxl.x86_64                    15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-gpu.x86_64             15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-gpu-gl.x86_64          15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-gpu-pci.x86_64         15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-gpu-pci-gl.x86_64      15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-vga.x86_64             15:7.0.0-2.el8                                            @raven-extras          
qemu-device-display-virtio-vga-gl.x86_64          15:7.0.0-2.el8                                            @raven-extras          
qemu-device-usb-host.x86_64                       15:7.0.0-2.el8                                            @raven-extras          
qemu-device-usb-redirect.x86_64                   15:7.0.0-2.el8                                            @raven-extras          
qemu-device-usb-smartcard.x86_64                  15:7.0.0-2.el8                                            @raven-extras          
qemu-img.x86_64                                   15:6.2.0-11.module+el8.6.0+1000+18e3b59f.5                @appstream             
qemu-kvm.x86_64                                   15:7.0.0-2.el8                                            @raven-extras          
qemu-system-arm.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-system-arm-core.x86_64                       15:7.0.0-2.el8                                            @raven-extras          
qemu-system-x86.x86_64                            15:7.0.0-2.el8                                            @raven-extras          
qemu-system-x86-core.x86_64                       15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-curses.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-egl-headless.x86_64                       15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-gtk.x86_64                                15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-opengl.x86_64                             15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-sdl.x86_64                                15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-spice-app.x86_64                          15:7.0.0-2.el8                                            @raven-extras          
qemu-ui-spice-core.x86_64                         15:7.0.0-2.el8                                            @raven-extras          
qemu-user.x86_64                                  15:7.0.0-2.el8                                            @raven-extras          
qemu-user-binfmt.x86_64                           15:7.0.0-2.el8                                            @raven-extras
```

## Debug Logs

Full log acquired with `PACKER_LOG=1` (`PACKER_LOG=1 sudo -E packer build boards/raspberry-pi/raspios-lite-arm.json`)

[packer.log](https://github.com/mkaczanowski/packer-builder-arm/files/10421070/packer.log)

csmithson12345, 2023-03-26:
i have run into this as well. 
ubuntu_server_20.04_arm64.json /hcl works
my updated ubuntu_server_22.04_arm64.json /hcl does not. i have updated the file url's to hit the 22.04 images.

ph20, 2024-01-01:
It's because the plugin can't find the registered `qemu-arm-static` binary.
This issue is related to the operating system specifics; in my case, on an Arch Linux host, it was fixed by installing the package qemu-user-static-binfmt, which contains the necessary orders for registering.
In another case, you might incorrectly set the `qemu_binary_source_path = "/usr/bin/qemu-arm-static"`.
I'm not sure, but I think it can be fixed by running this script: https://gist.github.com/yuanfang-chen/d3ad4959727bbab8682e

closed 2026-09-24 by dbast