mkaczanowski/packer-plugin-builder-arm #283: Is it still maintained?

michalfita, 2023-06-02:
@mkaczanowski & @dbast is this project still maintained? Maybe it's time to move it to organization and open to some other contributors? It think it gained enough interest to keep it useful (for example that packer plugin PR would require documentation change as well).

dbast, 2023-07-15:
Hi @michalfita. Would be happy to see it migrating to an org with more maintainers and doing the plugin step... etc. @mkaczanowski any objections? I can create an org and invite you both, if that is of help.

michalfita, 2023-07-15:
I've tried to reach @mkaczanowski even over his company's e-mail - no response unfortunately.

The proper plugin support would help avoiding Docker. Now I have image with this patch on my fork publicly available via ghcr.io registry.

dbast, 2023-07-15:
I also tried to reach him via mail twice regarding the repo rename required for the plug-in enablement... no reactions.

I have merge/push/release permissions, but cannot rename or move the repo to an org.

We could instead just create an org and push the repo history to an new repo and add hints to this then old one. A repo migration would have been awesome as that creates also a redirect from old to new location etc.

Any naming ideas for the org to be created?

michalfita, 2023-07-15:
Something in the spirit of `packer-cross` or `packer-embedded` would probably be best. The plugin itself should drop the `-arm` suffix as in future with different QEmu it could make RISC-V images as well.

If we can't reach @mkaczanowski soon, we can move one of the forks into organisation. Maybe attempt to phone his company would help I wonder.

dbast, 2023-07-17:
The new repo in the to be created org has to be named according to the plugin naming conventions https://developer.hashicorp.com/packer/docs/plugins/creation#naming-conventions which is `packer-plugin-NAME`. The org can be named the same or just without the `plugin` in it.

Dropping the `arm` suffix makes totally sense and I proposed to @mkaczanowski in December to rename to e.g. `packer-plugin-sbc` or `packer-plugin-qemu-chroot` ... but `packer-cross` and `packer-embedded` also make totally sense to me. ... some thoughts: 
* `qemu-chroot` is precisely describing what it does, but a bit too long for my taste
* `sbc` (single board computer) is the shortest and generic 
* `cross`: also short and generic (while it does not cross-compile, more cross-build)
* `embedded`: a bit longer .. the plugin can strictly also build images for not-embedded hardware .. but maybe the embedded folks find it easier to find it when named that way.

My personal preference order would be: sbc, embedded, cross.

The plugin does a bit more than just creating an image, but also creates partitions tables or extends existing images.

I just wrote him an eMail to make him aware about this thread. I am ready to create a Github org and also invite him there ... we can then either fork this repo or create an new repo from this template https://github.com/hashicorp/packer-plugin-scaffolding and merge the history of the current repo.

mkaczanowski, 2023-07-17:
@michalfita I have been explicitly ignoring your emails... I read your email and to me it seems **you have nothing to offer**, as in your message was pretty much: "Maybe, move it to organization" and that's it.

Why I think you have nothing to offer?:
https://github.com/mkaczanowski/packer-builder-arm/pulls?q=is%3Apr+michalfita+

That's just ^^ one reason. There are hundreds of people talking but there is a handful of people actually executing. And I am personally not interested in talking... If you want to help, do PR's.

I shared the repo permissions with @dbast because he is actively maintaining the plugin (**many many thanks!**). And if there were an org (which I am not opposed to at all), I would love to see who would be in that org in the first place. At this very moment the candidates are: me + @dbast, so ... does it make sense to create org for 2 people? IMHO no.

Therefore, if you have something to offer (other than talking), please make a PR. **PR's are always welcome!**

(this message is not intended to sounds harsh, but I get messages like this all over the place.)

mkaczanowski, 2023-07-17:
Regarding renaming the repo, sure that can get done. (yeah, I know I was postponing this for a while).

Closing the thread with a few answers:
1. Is it maintained? - somewhat, mainly by @dbast these days. 
2. Can the repo be ranamed? - yes it can
3. Will there be ownership passed to org? - not now, but if you (anyone) finds at least three **active** contributors I am willing to pass the ownership

closed 2023-07-17 by mkaczanowski

michalfita, 2023-07-18:
I'm sorry, @mkaczanowski, but with that attitude I withdraw all proposals I made and I'll stick with my fork I'm going to maintain for myself and my company. I did PR finally (when I had time to actually improve the code - I'm not Go programmer) that @dbast approved and merged in fact. My - as you called it - "talking" is result of being deeply interested in using your tool for serious purpose and help as far as I can, I don't care if I have any maintenance right. My fork has Docker image with my fixes that work for me - my company might be concerned that only I maintain that fork, but in the end we may write our own tooling if they don't trust this project.

What I find insulting in your post is the demand to do PRs while there is backlog of issues not addressed for quite some time. Does this project exists for its maintainers or its users? My believe was its the user base that counts, but it looks like I'm wrong.

michalfita, 2023-07-18:
> Dropping the arm suffix makes totally sense and I proposed to @mkaczanowski in December to rename to e.g. `packer-plugin-sbc` or `packer-plugin-qemu-chroot` ... but `packer-cross` and `packer-embedded` also make totally sense to me. ... some thoughts:

Do what you wish, @dbast, I wash my hands on carrying on - I'm going to maintain only my fork for myself from now on.