mkaczanowski/packer-plugin-builder-arm #341: No space left on device while trying to install some packages

MiroYld, 2024-04-04:
I have a problem with my packer configuration.
I use the shell type provisioner to install the packages via scripts but I have a problem with the allocated disk space
my builder section
```
            "image_size": "2GB",
            "image_type": "dos",
            "image_partitions": [
                {
                    "name": "boot",
                    "type": "c",
                    "start_sector": "8192",
                    "filesystem": "vfat",
                    "size": "256M",
                    "mountpoint": "/boot"
                },
                {
                    "name": "root",
                    "type": "83",
                    "start_sector": "532480",
                    "filesystem": "ext4",
                    "size": "5G",
                    "mountpoint": "/"
                }
            ],
```
I'm assigning 5G so I don't really understand how it's possible

In my provisioner section I just do
```
        {
            "type": "shell",
            "script": "./provisioning-scripts/install-packages.sh"
        }
```
And my script is very simple: 
```bash
#!/bin/bash

sudo apt-get update -y && sudo apt-upgrade -y
sudo apt-get install -y \
    autoconf \
    autoconf-archive \
    automake \
    build-essential \
    ca-certificates \
    evtest \
    jq \
    less \
    libnfc-dev \
    libnfc6 \
    libpcsclite-dev \
    libpcsclite1 \
    libpigpiod-if-dev \
    libpigpiod-if2-1 \
    libqt5gui5 \
    libqt5multimedia5 \
    libqt5serialport5 \
    libqt5serialport5-dev \
    libqt5sql5 \
    libqt5sql5-sqlite \
    libqt5websockets5 \
    libqt5websockets5-dev \
    libtool \
    openssl \
    pcscd \
    procps \
    pulseaudio \
    qtbase5-dev \
    qtbase5-dev-tools \
    sox \
    sqlite3 \
    strace \
    udev \
    unzip \
    wget
```
Here is some logs:
 ```
    arm: tar: ./symbols: Cannot write: No space left on device
    arm: tar: ./triggers: Cannot write: No space left on device
    arm: tar: Exiting with failure status due to previous errors
    arm: dpkg-deb: error: tar subprocess returned error exit status 2
    arm: dpkg: error processing archive /tmp/apt-dpkg-install-3V5xur/187-libqt5widgets5_5.15.8+dfsg-11_arm64.deb (--unpack):
    arm:  dpkg-deb --control subprocess returned error exit status 2
    arm: tar: ./md5sums: Cannot write: No space left on device
    arm: tar: ./shlibs: Cannot write: No space left on device
    arm: tar: ./symbols: Cannot write: No space left on device
    arm: tar: ./triggers: Cannot write: No space left on device
    arm: tar: Exiting with failure status due to previous errors
    arm: dpkg-deb: error: tar subprocess returned error exit status 2
    arm: dpkg: error processing archive /tmp/apt-dpkg-install-3V5xur/188-libqt5opengl5_5.15.8+dfsg-11_arm64.deb (--unpack):
    arm:  dpkg-deb --control subprocess returned error exit status 2
    arm: tar: ./md5sums: Cannot write: No space left on device
    arm: tar: ./shlibs: Cannot write: No space left on device
    arm: tar: ./symbols: Cannot write: No space left on device
    arm: tar: ./triggers: Cannot write: No space left on device
    arm: tar: Exiting with failure status due to previous errors
    arm: dpkg-deb: error: tar subprocess returned error exit status 2
    arm: dpkg: error processing archive /tmp/apt-dpkg-install-3V5xur/189-libqt5printsupport5_5.15.8+dfsg-11_arm64.deb (--unpack):
    arm:  dpkg-deb --control subprocess returned error exit status 2
    arm: dpkg: too many errors, stopping
    arm: dpkg: error: failed to write status database stanza about 'apparmor' to '/var/lib/dpkg/status': No space left on device
==> arm: E: Sub-process /usr/bin/dpkg returned an error code (2)
```

farazmd, 2024-04-13:
The top level `image_size` key needs to be set to the desired size and `image_build_method` needs to be set to `resize`. The `size` key within the `image_partitions` for the root partition needs to be set to `0`.

More info and instructions with examples are here - https://github.com/mkaczanowski/packer-builder-arm?tab=readme-ov-file#resizing-image

MiroYld, 2024-04-16:
Thank you for your response, I tried this 
```
            "image_path": "pi4-bookworm.img",
            "image_size": "500M",
	    "image_build_method": "resize",
            "image_partitions": [
                {
                    "name": "boot",
                    "type": "c",
                    "start_sector": "8192",
                    "filesystem": "vfat",
                    "size": "256M",
                    "mountpoint": "/boot"
                },
                {
                    "name": "root",
                    "type": "83",
                    "start_sector": "532480",
                    "filesystem": "ext4",
                    "size": "0",
                    "mountpoint": "/"
                }
            ],
``` 
But I have an error with the image resizing 
```
    arm: resizing the image file pi4-bookworm.img to 500M
==> arm: error while resizing the image exit status 1: WARNING: Image format was not specified for 'pi4-bookworm.img' and probing guessed raw.
==> arm:          Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
==> arm:          Specify the 'raw' format explicitly to remove the restrictions.
Build 'arm' errored after 22 seconds 270 milliseconds: build was halted
```
And I have the same error when I try to build : https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi/raspbian-resize.json

farazmd, 2024-04-22:
I think the error is because the requested size is smaller than the current image size `500M`. I haven't tried resizing it to a smaller size, but I think you need to build the image from scratch if you need a smaller image,otherwise you'd lose the rootfs. I see this warning right after the error :
```shell
 qemu-img: warning: Shrinking an image will delete all data beyond the shrunken images end.
 Before performing such an operation, make sure there is no important data there.

```

 Look at the archlinux example here for building from scratch - https://github.com/mkaczanowski/packer-builder-arm/blob/master/boards/raspberry-pi/archlinuxarm.json#L10. Might have to look around for a debian example.

closed 2026-09-24 by dbast